PX4/PX4-Autopilot #5500: 3dr solo sensor init isn't working

lovettchris, 2016-09-16:
The following code does not work in ~/ROMFS/px4fmu_common/init.d/rc.sensors:

	# external MPU6K is rotated 180 degrees yaw
	if mpu6000 -X -R 4 start
	then
		echo "Found BOARD_FMUV3"
		set BOARD_FMUV3 true
	else
		echo "Setting BOARD_FMUV3 anyway"
		set BOARD_FMUV3 true
	fi

I added the "echo" statements for debugging and it prints "Setting BOARD_FMUV3 anyway", meaning the "mpu6000" task did not return true, even though it did actually start.  I know it started because the PX4 system console prints "MPU6000 on SPI bus 1 at 4 (1000 KHz)".  So I think something is wrong with how the task returns true or false.  When I compare it with other drivers I see a difference in exit(0) versus exit(1).  mpu6000 seems to always call exit(1), I think...

bkueng, 2016-09-19:
According to https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.cpp#L2330
the driver does return a proper error code. However this still does not explain your observed behavior.
It could be that the device initialization succeeds (https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.cpp#L2266), and prints the message, but fails afterwards.
Can you debug this further?

LorenzMeier, 2016-12-26:
This should be fixed now.

closed 2016-12-26 by LorenzMeier