PX4/PX4-Autopilot #5501: Reboot with forcing to bootloader gets swallowed by MAVLink

AndreasAntener, 2016-09-16:
@LorenzMeier why are we using `MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN` with `param1 == 3` as link shutdown? https://github.com/PX4/Firmware/blob/7d7e40f59473ed538c61f8b76020ca505ee52487/src/modules/mavlink/mavlink_receiver.cpp#L353

According to the specs that should do an FC reboot with forcing it to bootloader. Which is implemented but never happens because of the above logic.

Funnily this could have been the same issue: https://github.com/PX4/Firmware/pull/5473 (although normal boot was ok for him I guess)

LorenzMeier, 2016-09-16:
This is indeed a case where we had a custom use and then the spec got extended.

LorenzMeier, 2016-09-16:
Closed in d98bbd9.

closed 2016-09-16 by LorenzMeier