PX4/PX4-Autopilot #5508: SITL gazebo ROS issues

eric1221bday, 2016-09-19:
So I've been having some issues running the gazebo ros simulation specifically with the iris opt flow model. Everything works fine and the simulation and mavros both starts. However despite the fact that there are plugins for both the px4flow and the lidar for gazebo that are suppose to publish ros messages, only the camera feed is actually publishing to ros. In addition if I look into the specific messages in gazebo for both the flow and the lidar, gazebo will say that it cannot parse the message. I suspect it has something to do with the fact that gazebo giu cannot access the custom plugins in the mavlink_gazebo package, thus only the default camera plugin works, but does anyone have any insights?

darknight-007, 2016-09-28:
I think I am facing the same issue. With an earlier version (July), I could run 
make no_sim posix_sitl_default gazebo 

...followed by (in a new terminal), roslaunch gazebo_ros empty.world

and then on gzclient, insert the 3DR Iris (for example), and see the "flight simulator" connect to the PX4 simulator. And then, I could do "commander takeoff" etc, or make use of ROS camera plugins. 

On current master, the PX4 simulator continues to wait with the following message:
"Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed"

As a side note, the port number (14560) did not show up in the old no_sim mode.

michalgnat, 2016-11-02:
Hi darknight-007, I'm fairly new to linux and I don't understand how to create symbolic links to Firmware and sitl_gazebo folders in the catkin_ws folder.

What I undestood is to $ln -s the Firmware folder into the sitl_gazebo package and then $ln -s the sitl_gazebo folder into the Firmware/Tooks/sitl_gazebo folder.

Can you please print out the linux commands you did to make it more clear?

Jaeyoung-Lim, 2016-11-02:
I am having the same problem
```
no_sim=1 make posix_sitl_default gazebo
```
has been working well before but the current master is not working.

@darknight-007  Could you be more specific on solving the problem?

closed 2017-04-03 by mhkabir

Stoven-Dubois, 2017-10-05:
Hello, by following the same tutorial (http://dev.px4.io/simulation-ros-interface.html#launching-gazebo-with-ros-wrappers), I still can not get the topics to be published in ROS.
If I use the launch file as advised:

roslaunch px4 mavros_posix_sitl.launch

I can get the topics to exist, but not to have them be published.

I also tried the manual version: 

no_sim=1 make posix_sitl_default gazebo
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default
roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/iris.world

but this only gives me: 

INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

YuYangguang, 2017-10-13:
I follow your guide and create a link, then catkin_make.I  But still can not get the topics to be published in ROS. The phenomenon is just the same as @Stoven-Dubois . Did you do other operations? @darknight-007 . And did you solve this issue? @Stoven-Dubois

Stoven-Dubois, 2017-10-16:
No, sorry @YuYangguang. I did try other things, but none of them worked.

YuYangguang, 2017-10-16:
I have solved this issue, please see here   [https://github.com/PX4/Firmware/issues/8088#issuecomment-336740438](url)， you'd better to update your firmware to the latest. After I updated my firmware, many issues was addressed.