PX4/PX4-Autopilot #5511: Corner case VTOL_LAND

sanderux, 2016-09-19:
Presumably: the vtol came out of back transition directly on top of the waypoint. It therefor skipped over the navigation step in the sequence and moved directly to land mode while still having a lot of velocity.

purple: setpoint
red: actual

[image]

http://logs.uaventure.com/view/yVa84XvfcDNq85UrEcRpuG

sanderux, 2016-09-19:
Waiting until @AndreasAntener syncs his stuff, after that i can look if / how this could be fixed and if that would still be needed

sanderux, 2016-09-19:
google earth flight
[flight.kmz.tar.gz](https://github.com/PX4/Firmware/files/480759/flight.kmz.tar.gz)

tubeme, 2016-09-22:
For the back transition our experience in missions is that the plane always makes the transition too late and always misses its landing point. Then in hover it starts to go back. But you know VTOL and backward movement is not the best combination. It is always better to have the transition way before the landing spot and then go forward in hover until on the spot. We have to take in to account the speed of the plane and its wight in order to calculate properly where the transition should precede the landing spot in mission.

sanderux, 2016-09-23:
@tubeme Until this is addressed (likely i would calculate distance based on groundspeed) I would recommend using a regular transition command if you want it not to overshoot.

closed 2017-01-20 by sanderux