PX4/PX4-Autopilot #5514: Non intuitive sensor calibration failure

CarlOlsson, 2016-09-20:
I just remounted the Pixracer and I could not calibrate the acc since I only got "error, invalid orientation". I was only able to calibrate after I manually reset every parameter listed below to default values through QGC. 
CAL_ACC0_XOFF
CAL_ACC0_XSCALE
CAL_ACC0_YOFF
CAL_ACC0_YSCALE
CAL_ACC0_ZOFF
CAL_ACC0_ZSCALE
(Thanks @sverling for the tip)

mhkabir, 2017-04-03:
Still an issue?

CarlOlsson, 2017-04-03:
I don't know if this is fixed. To reproduce one would need to set up a model, set IMU orientation, calibrate, rotate the IMU, and now try to set up the plane again. This was not possible before without manually resetting the cal in QGC. I will keep in mind to test it again when I have time

mhkabir, 2017-04-03:
Cool. Let us know if it still happens.

closed 2017-04-12 by dagar