PX4/PX4-Autopilot #5515: error building firmware for crazyflie 2.0

sneiman, 2016-09-20:
Noob with px4. Trying to reflash Crazyflie 2.0. No problem with Bootloader. But cannot build crazyflie_default. Here is the error after issuing make crazyflie_default upload:

[ 59%] Building CXX object src/modules/syslink/CMakeFiles/modules__syslink.dir/syslink_main.cpp.obj
[ 60%] Building CXX object src/modules/attitude_estimator_q/CMakeFiles/modules__attitude_estimator_q.dir/attitude_estimator_q_main.cpp.obj
[ 60%] Building CXX object src/modules/navigator/CMakeFiles/modules__navigator.dir/geofence.cpp.obj
[ 60%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_orb_subscription.cpp.obj
/home/seth/dev/px4/src/Firmware/src/modules/syslink/syslink_main.cpp: In member function 'void Syslink::handle_raw(syslink_message_t*)':
/home/seth/dev/px4/src/Firmware/src/modules/syslink/syslink_main.cpp:490:6: error: 'struct input_rc_s' has no member named 'timestamp_publication'
   rc.timestamp_publication = hrt_absolute_time();

bkueng, 2016-09-20:
Indeed, the field `timestamp_publication` got removed very recently, fixed in #5517

sneiman, 2016-09-20:
That was fast! Thanks – new to this repo, how long before these are likely to be merged?

S

This entire message is confidential. If it isn't intended for you, you may not use it – so please throw it away and forget about it.

bkueng, 2016-09-20:
Usually someone else will review it, but for such a small change, I'll merge when CI passes.

sneiman, 2016-09-20:
Thanks.

This entire message is confidential. If it isn't intended for you, you may not use it – so please throw it away and forget about it.

bkueng, 2016-09-20:
Merged!

closed 2016-09-20 by bkueng

sneiman, 2016-09-20:
Thanks for headsup. Builds successfully – wont download, but I’m working on  it!