PX4/PX4-Autopilot #5524: px4fmu-v2_default compilation error : nuttx-export.zip is missing

VictorLSogilis, 2016-09-21:
Hi,

I'm following the build instructions from the developper documentation and I encounter a build issue about nuttx-export.zip missing. Here is the error log : 

[error_log.txt](https://github.com/PX4/Firmware/files/484437/error_log.txt)

I'm building from a Debian in this configuration : 
```

lambret@debian-sogilis:~/ibubble/Firmware2$ git log -1
commit 392391e8df34de5590dde9328be4c7cd02403bf9
Author: Nacho Carnicero <i.carnicerotresca@gmail.com>
Date:   Mon Sep 19 17:15:03 2016 +0200

    Fix VTOL status in forward transition for tiltrotor models
lambret@debian-sogilis:~/ibubble/Firmware2$ cd NuttX/
lambret@debian-sogilis:~/ibubble/Firmware2/NuttX$ git log -1
commit 93df87120129e90f6d1a71ecabdcc7fde3e6bb4b
Author: David Sidrane <david_s5@nscdg.com>
Date:   Thu Aug 25 05:06:22 2016 -1000

    Fix Merge and really remove delay
lambret@debian-sogilis:~/ibubble/Firmware2/NuttX$ cmake --version
cmake version 3.3.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

I suppose the missing nuttx-export.zip should be created by nuttx build but I dont understand how it's working and why it's not generated.

I would be happy to provide any information needed to investigate the problem and of course if I have a solution that I understand I will submit a patch :)

davids5, 2016-09-21:
Did you run ```make clean``` first?

VictorLSogilis, 2016-09-21:
I tried to make clean and make distclean before building without success.

Finally I installed a fresh 16.04 Ubuntu on a Virtual Machine and it build without error. I suspect the problem came from a package difference somewhere but I'm not sure how I can investigate that.

julianoes, 2016-10-11:
I would suggest to compare the versions of all the tools which are needed to build the firmware. However, since you were able to workaround the issue using the Ubuntu VM, I'll close this issue.

closed 2016-10-11 by julianoes