PX4/PX4-Autopilot #5525: Building errors with px4fmu-v2_default

alpha17, 2016-09-21:
As I'm trying to build an example like the one on the development guide, I got some problem when using the make px4fmu-v2_default instruction. The errors are emphasized in the console window as you can see below:
[image]

Using Windows as my OS. Very new to PX4. Hope this isn't a stupid question. But I just can solve it by myself.  I will be grateful for any help from anyone.

julianoes, 2016-09-27:
Hi @alpha17, have you seen the note that Windows is not officially supported here:
http://dev.px4.io/starting-installing-windows.html

I would recommend to try it on a Linux in VirtualBox or try Bash On Windows.

dagar, 2016-09-27:
@LorenzMeier @julianoes it's going to be annoying, but realistically we need a single windows build or it's just going to keep breaking all the time.

alpha17, 2016-09-28:
Thanks for your advice and adjustment on the title, @julianoes! I did read that part in the devguide before. But I thought I should give it a try on Windows first. Will get a Linux OS installed in these couple days and see if there's any problem.

alpha17, 2016-10-02:
Built the project without error using Ubuntu. Thanks again,  @julianoes!

julianoes, 2016-10-02:
Cool, thanks for the feedback!

closed 2016-10-02 by julianoes