PX4/PX4-Autopilot #5526: Total loss of control, apparently due to PX4_ERR

ndepal, 2016-09-21:
I have repeatedly experienced a total loss of control/crashes with my quad after making some minor changes to the position controller.
I have narrowed down the problem to a print statement. 

**Problem**
The crash can go one of two ways:
1) The quad's rotors stop spinning mid-air, as if it was disarmed.
2) The quad becomes entirely unresponsive. Communication with QGC (through a RPi companion computer) is lost. Control via the RC is also lost. I am not able to change modes. Since the print statement I have added is in `MulticopterPositionControl::control_auto`, the problem occurs when I am in mission mode and I am thus completely unable to control the quad in any way. Position hold performance in mission loiter (the assumed state of the quad) is poor and becomes worse over time.

The latter of the two outcomes is of course far worse. I was forced to watch the quad as it circled around in the air with increasing radius and speed until the battery died and the quad crashed from a great altitude.

Symptoms prior to the crash/loss of communication are reports of mag or baro failure.
Looking at the logs (which are all corrupt as I have no way of disarming), there are large gaps in all logged data. I would guess they appear around the same time as the sensor failure messages.

**How to reproduce**
Through various tests and a string tied to the quad, I have narrowed down the problem to a simple  `PX4_ERR` statement in the control loop. With the following lines added to the top of `MulticopterPositionControl::control_auto`, based on master as of d98bbd9b825f3610b6c0dad6e2274ca3de5ab639, the issue can be reproduced:

```
static int printcnt = 0;
if (printcnt % 10 == 0) {
	PX4_ERR("control_auto %d, %f %f %f", printcnt, M_PI, 1.4, 1e-7);
}
printcnt++;
```

Note that a simpler print statement with fewer arguments does not produce the issues for me.

**Logs**
Logging cuts out entirely when the quad becomes unresponsive so the logs don't provide too much insight. Nevertheless, here is a log of a test flight where the quad dropped out of the sky:
http://logs.uaventure.com/view/vU58fzj6GYmeMJrHXRQyPJ

The corresponding messages are:
```
2016_09_20_15_03_20: ARMED by RC
2016_09_20_15_03_20: [blackbox] /fs/microsd/log/2016-09-20
2016_09_20_15_03_20: [blackbox] recording: 15_03_20.px4log
2016_09_20_15_03_21: Takeoff detected
2016_09_20_15_03_24: [lpe] lidar init: mean 52 cm stddev 19 cm
2016_09_20_15_03_25: [lpe] flow init: quality 233 std 32
2016_09_20_15_03_31: takeoff to 3.0 meters above home
2016_09_20_15_03_38: [lpe] lidar fault,  beta 54.76
2016_09_20_15_03_38: [lpe] lidar fault,  beta 27.02
2016_09_20_15_03_42: mission finished, loitering
2016_09_20_15_04_01: Executing mission
2016_09_20_15_04_06: Mag #0 failover : Data timeout!
2016_09_20_15_04_06: [lpe] flow timeout 
2016_09_20_15_04_06: [lpe] lidar timeout 
2016_09_20_15_04_07: [lpe] lidar init: mean 247 cm stddev 8 cm
2016_09_20_15_04_07: [lpe] flow init: quality 254 std 0
2016_09_20_15_04_07: [lpe] lidar fault,  beta  9.27
2016_09_20_15_04_08: [lpe] lidar fault,  beta 17.69
2016_09_20_15_04_09: [lpe] lidar fault,  beta 76.92
2016_09_20_15_04_09: [lpe] lidar fault,  beta 25.79
2016_09_20_15_04_12: mission finished, loitering
2016_09_20_15_04_32: Executing mission
2016_09_20_15_04_34: [lpe] lidar fault,  beta 194.74
2016_09_20_15_04_36: [lpe] lidar fault,  beta 12.63
2016_09_20_15_04_40: Mag #0 failover : Data timeout!
2016_09_20_15_04_41: mission finished, loitering
2016_09_20_15_04_51: no valid mission available, loitering
2016_09_20_15_05_02: Executing mission
2016_09_20_15_05_10: mission finished, loitering
2016_09_20_15_05_19: Executing mission
2016_09_20_15_05_27: mission finished, loitering
2016_09_20_15_05_39: no valid mission available, loitering
2016_09_20_15_05_43: Executing mission
2016_09_20_15_05_57: Mag #0 failover : Data timeout!
2016_09_20_15_05_57: [lpe] flow timeout 
2016_09_20_15_05_57: [lpe] lidar timeout 
2016_09_20_15_05_57: [lpe] baro fault, r 36.08 m, beta 143.99
2016_09_20_15_05_58: [lpe] lidar init: mean 265 cm stddev 12 cm
2016_09_20_15_05_58: [lpe] lidar fault,  beta 14.45
2016_09_20_15_05_58: [lpe] flow init: quality 252 std 2
2016_09_20_15_06_00: Mag #0 failover : Data timeout!
2016_09_20_15_06_00: [lpe] flow timeout 
2016_09_20_15_06_00: [lpe] lidar timeout 
2016_09_20_15_06_01: [lpe] lidar init: mean 291 cm stddev 4 cm
2016_09_20_15_06_01: [lpe] flow init: quality 239 std 47
2016_09_20_15_06_01: [lpe] lidar fault,  beta 38.60
2016_09_20_15_06_02: mission finished, loitering
2016_09_20_15_06_32: Executing mission
2016_09_20_15_06_32: takeoff to 3.0 meters above home
2016_09_20_15_06_36: Mag #0 failover : Data timeout!
2016_09_20_15_06_36: [lpe] flow timeout 
2016_09_20_15_06_36: [lpe] lidar timeout 
2016_09_20_15_06_37: [lpe] lidar init: mean 140 cm stddev 0 cm
2016_09_20_15_06_37: [lpe] flow init: quality 242 std 40
2016_09_20_15_06_39: [lpe] lidar fault,  beta 63.88
2016_09_20_15_06_39: [lpe] lidar fault,  beta 23.57
2016_09_20_15_06_40: Mag #0 failover : Data timeout!
2016_09_20_15_06_42: mission finished, loitering
2016_09_20_15_07_05: Executing mission
2016_09_20_15_07_08: Mag #0 failover : Data timeout!
2016_09_20_15_07_08: [lpe] flow timeout 
2016_09_20_15_07_09: [lpe] lidar timeout 
2016_09_20_15_07_09: [lpe] lidar init: mean 150 cm stddev 10 cm
2016_09_20_15_07_09: [lpe] flow init: quality 239 std 50
2016_09_20_15_07_10: [lpe] lidar fault,  beta 50.66
2016_09_20_15_07_11: [lpe] lidar fault,  beta 23.65
2016_09_20_15_07_14: mission finished, loitering
```

I am flying with a PixFalcon with the firmware based on the above mentioned master. I am using LPE and sdlog2.

I could remove those print statements and move on as I only put them in for some tests in SITL, but I do feel that the ramifications of such an innocuous change are quite severe and should be addressed.

LorenzMeier, 2016-09-21:
You have a stack overflow. You need to increase the stack if you add print commands

closed 2016-09-21 by LorenzMeier

ndepal, 2016-09-21:
Thanks for the response. How can I prevent running into this type of problem in the future? How do I check the stack size vs max allowable size?

tumbili, 2016-09-21:
@ndepal The "top" command in the shell will show you stack usage for all processes.
Generally when you made a considerable amount of changes to the code you should check if the process has some stack free. Consider also that you might have functions which consume lots of stack but which will only be called after a certain event.
Also you can use the "free" command to check how much memory you have left in general.

ndepal, 2016-09-21:
Thanks @tumbili !