PX4/PX4-Autopilot #5527: Error building posix_sitl_default

nacho-carnicero, 2016-09-21:
Hi, 

I'm getting an error while building PX4 for SITL simulation with Gazebo. I installed the pre-requisites as stated in [this page](http://dev.px4.io/simulation-gazebo.html) and ran `make posix_sitl_default gazebo`. I got the following error:
```
--   Found protobuf, version 3.0.2
Gazebo version: 7.3
CMake Error at CMakeLists.txt:80 (message):
  protobuf version: 3.0.2 not compatible, must be >= 2.5.0 and < 3.0.0

-- Configuring incomplete, errors occurred!
```

So it appears that the protobuf version I have is too new, therefore I downloaded the newest version below v3.0.0 (as the error log is suggesting) which is v2.6.1 and I installed it. Then I proceeded to re-run the build as follows:

```
make posix_sitl_default clean
make posix_sitl_default gazebo
```

I get a new error which is very long, but I'll only display the beginning since it says where it does come from:

```
In file included from SensorImu.pb.cc:5:
In file included from ./SensorImu.pb.h:27:
/usr/local/Cellar/gazebo7/7.3.1_4/include/gazebo-7/gazebo/msgs/quaternion.pb.h:12:2: error: This file was generated by a newer version of protoc which is
#error This file was generated by a newer version of protoc which is
 ^
/usr/local/Cellar/gazebo7/7.3.1_4/include/gazebo-7/gazebo/msgs/quaternion.pb.h:13:2: error: incompatible with your Protocol Buffer headers.  Please update
#error incompatible with your Protocol Buffer headers.  Please update
 ^
/usr/local/Cellar/gazebo7/7.3.1_4/include/gazebo-7/gazebo/msgs/quaternion.pb.h:14:2: error: your headers.
#error your headers.
 ^
```
So now it seems like some headers that are used by protobuf has been generated by a newer version, and thus it is not able to recognize them. The problem is that there is no other version in [the releases of protobuf page](https://github.com/google/protobuf/releases?after=v3.0.0-alpha-1) so I don't know where the problem comes from.

I'm building all of this in a Mac with gazebo7 installed today and with the latest version of the master firmware.

P.S: If I change the CMakeLists.txt in `Tools/sitl_gazebo` and allow for having protobuf versions above 3.0.0 the simulation executes but without the world.

bkueng, 2016-09-22:
> If I change the CMakeLists.txt in Tools/sitl_gazebo and allow for having protobuf versions above 3.0.0 the simulation executes but without the world

What do you mean exactly? If you see the quad on a ground plane, it's actually correct. There are no buiildings in this world.

I think @jgoppert could explain why the `protobuf < 3.0.0` requirement was added.

jgoppert, 2016-09-23:
@bkueng I couldn't get the code to build without protobuf < 3.0.0.  @nacho-carnicero  2.6.1 is the correct version, can you try deleting all build files and trying again. Also, try using gazebo 7.0.0:

gazebo --version
Gazebo multi-robot simulator, version 7.0.0

It looks like you are using a newer gazebo version, which may now use the 3.0.0 protobuf.

nacho-carnicero, 2016-09-23:
Ok @bkueng I got confused because I saw in the video of the dev page all the environment I was not seeing in my simulation, therefore the build worked.

@jgoppert  yes, I am using Gazebo 7.3.1. Maybe we could add a check on the CMakeLists in order to allow using higher versions of protobuf if the version of Gazebo is above 7.0.0. Or do you recommend downgrading to v7.0.0 for other compatibility issues?

julianoes, 2016-09-30:
For now as a workaround on Mac I found that switching back to the old version works:

```
brew switch protobuf 2.6.1
```

julianoes, 2016-09-30:
I've updated the instructions how to get to the older protobuf version in brew, see https://github.com/PX4/Devguide/pull/45.

julianoes, 2016-10-11:
I've reverted this but added a PR here for it:
https://github.com/PX4/sitl_gazebo/pull/53

If anyone could cross-test that would be great.

julianoes, 2016-10-12:
https://github.com/PX4/sitl_gazebo/pull/53 has gone in, this should be fixed now.

closed 2016-10-12 by julianoes