PX4/PX4-Autopilot #5535: Spektrum RC doesn't work

VictorSu2, 2016-09-23:
Hi,
I just followed the instructions here "Building PX4 with the FC_ADDON drivers" to make the Spektrum work with px4 https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md
But the the radio part on QGC shows nothing when i move the joysticks while all the other calibration passes.
I tried several times and i promise that the Spektrum controller and receiver are binded, snapdragon and QGC are in the same wifi. 

Here is the  mini-dm output:

> Running mini-dm version: 3.0
> Device found with Product ID 0x9025. Continuing...
> mini-dm is waiting for a DMSS connection...
> DMSS is connected. Running mini-dm...
> [08500/00]  29:54.040  HAP:61:In dspal_entry  0214  main.cpp
> [08500/00]  29:54.270  HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config  0178  main.cpp
> [08500/00]  29:54.270  HAP:61:reading commands from /dev/fs/px4.config   0185  main.cpp
> [08500/02]  29:56.285  HAP:61:Error: Parameter UART_ESC_BAUDRTE not found.  0470  param.c
> [08500/02]  29:56.441  HAP:61:Sleeping for 1, 1000000  0118  apps.h
> [08500/00]  29:57.545  HAP:61:3504116 Reset MPU9250   0089  MPU9250.cpp
> [08500/00]  29:57.608  HAP:60:3568353 IMU temperature initialized to: 57.692520   0516  MPU9250.cpp
> [08500/00]  29:57.610  HAP:61:3570904 additional sensor configuration succeeded   0187  BMP280.cpp
> [08500/02]  29:57.636  HAP:20527:LED: getHandle fail   0275  commander_helper.cpp
> [08500/02]  29:57.636  HAP:20527:LED init failed  1292  commander.cpp
> [08500/02]  29:57.636  HAP:20527:Buzzer: px4_open fail   0138  commander_helper.cpp
> [08500/02]  29:57.636  HAP:20527:Buzzer init failed  1296  commander.cpp
> [08500/02]  29:57.638  HAP:20527:Preflight checks always pass on Snapdragon.  0392  PreflightCheck.cpp
> [08500/02]  29:57.708  HAP:61:Sleeping for 1, 1000000  0118  apps.h
> [08500/00]  29:58.708  HAP:61:PX4 Devices:  0539  vdev.cpp
> [08500/00]  29:58.708  HAP:61:DF Devices:  0552  vdev.cpp
> [08500/00]  29:58.708  HAP:61:   /dev/imu0  0562  vdev.cpp
> [08500/00]  29:58.708  HAP:61:   /dev/baro0  0562  vdev.cpp
> [08500/00]  29:58.708  HAP:61:Files:  0588  vdev.cpp
> [08500/00]  29:58.708  HAP:61:Active Tasks:  0314  px4_qurt_tasks.cpp
> [08500/00]  29:58.708  HAP:61:   dspal      2097663  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.708  HAP:61:   hpwork     2097661  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   lpwork     2097660  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   wkr_hrt    2097659  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   qshell     2097657  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   sensors    2097656  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   commander  2327034  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   ekf2       2154999  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   mc_pos_control 2580981  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:   mc_att_control 2097651  0318  px4_qurt_tasks.cpp
> [08500/00]  29:58.710  HAP:61:Devices:  0572  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/_obj_  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/qshell_req0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/sensor_accel0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/sensor_gyro0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/parameter_update0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/sensor_mag0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/input_rc0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/differential_pressure0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_control_mode0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/rc_parameter_map0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/manual_control_setpoint0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/actuator_controls_00  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/sensor_baro0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/sensor_combined0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_status0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/actuator_armed0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/safety0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/mission_result0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/geofence_result0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/offboard_control_mode0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_global_position0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_local_position0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_attitude0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_land_detected0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_gps_position0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_command0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/battery_status0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/subsystem_info0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/position_setpoint_triplet0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/system_power0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vtol_vehicle_status0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/cpuload0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/mavlink_log0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/commander_state0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/airspeed0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/optical_flow0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/distance_sensor0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vision_position_estimate0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/estimator_status0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/wind_estimate0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/ekf2_innovations0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/control_state0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_attitude_setpoint0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_local_position_setpoint0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_global_velocity_setpoint0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/vehicle_rates_setpoint0  0578  vdev.cpp
> [08500/00]  29:58.710  HAP:61:   /obj/multirotor_motor_limits0  0578  vdev.cpp
> [08500/00]  29:59.303  HAP:12333:EKF using pressure height - commencing alignment  0407  ekf.cpp
> [08500/00]  30:04.391  HAP:12333:EKF alignment complete  0064  control.cpp
> [08500/02]  30:40.623  HAP:16430:Preflight checks always pass on Snapdragon.  0392  PreflightCheck.cpp

Here is the shell output:

> 
> root@linaro-developer:/home/linaro# ./px4 mainapp.config 
> sh: 1: cannot create /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor: Directory nonexistent
> data path: .
> commands file: mainapp.config
> 
> ______  __   __    ___ 
> | ___ \ \ \ / /   /   |
> | |_/ /  \ V /   / /| |
> |  __/   /   \  / /_| |
> | |     / /^\ \ \___  |
> \_|     \/   \/     |_/
> 
> px4 starting.
> 
> INFO  [mavlink] mode: Normal, data rate: 2880 B/s on udp port 14556 remote port 14550
> pxh> INFO  [mavlink] using network interface wlan0, IP: 192.168.1.248
> INFO  [mavlink] with broadcast IP: 192.168.1.255
> INFO  [mavlink] partner IP: 192.168.1.155

Can anyone tell me what is wrong with the snapdragon set up?

Thanks

LorenzMeier, 2016-09-27:
@bharath374 Could you have a look?

LorenzMeier, 2016-10-23:
@bharath374 We will replace the Qualcomm driver with one based off src/drivers/px4fmu.

LorenzMeier, 2016-12-26:
We did replace the driver since then, will be in master soon.

closed 2016-12-26 by LorenzMeier