PX4/PX4-Autopilot #5541: UGENT!!!Yaw angle IMU drift and  the strange behavior of yaw output channel

zer1988, 2016-09-24:
Hi LorenzMeier  and every body,
Currentlly, I can control well my drone's roll and pitch angle using my own PID code. However, a big trouble occurs when I start to tune yaw angle.
For yaw angle, when I put my drone one the floor or in the process of heading stablity, the value of yaw angle changes 0.1 degree/ 0.2 seconds. Therefore, After 7 or 10 seconds of my flight, my drone will rotate counter clock-wise.
After that, I changed the sensor for yaw angle using Vicon system so that my drone no longer have a drift at the yaw angle input. My problem now is that if I set up the yaw channel output is 0, my drone will automatically rotate counter-clockwise when taking- off. All of my efforts now whilst tuning yaw angle using Kp of PID seems to resist this rotation. 
Anybody can explain me this situation and guide me how to fix it or how I can reduce the drift of yaw angle.
Thanks,

LorenzMeier, 2016-09-24:
Sorry, but you're not even sharing your custom PID code. I can't see how we can help you discussing a mistake in something that we can't even look at. Besides, this issue tracker is for issues in the PX4 flight control code - and very obviously the issue you're having with your own does not belong here.

We're happy to help you with issues related that are to any code in this repository or help you contribute to it.

closed 2016-09-24 by LorenzMeier

zer1988, 2016-09-24:
Hi LorenzMeier,
Sorry to miss my log files here. On Tuesday I will take some and post here for IMU yaw  angle inside pixhawk. 

Going back to the last task with Vicon system, I used mavros/actuator and produces zero at yaw output channel. Unfortunately, it rotates anti-lockwise. My PID is too simple like I used VICON to get the current yaw angle. the error between yaw setpoint and yaw angle is the error. This value will be multiplied by Kp. IMU inside pixhawk provides the yaw rate angle and this value will be multiplied by kd. Ki is ignored. 
I will capture some photos on my screen on Tuesday and discuss deeper with you. Is it ok, LorenzMeier?
Thanks,