PX4/PX4-Autopilot #5542: bug wwhen using LPE

JohnSnowball, 2016-09-25:
https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L392

I encountered a serious bug when using LPE, during transition from mc -> fw, the whole system is blocked, I tried command "top" in console, it's also dead, nothing showed but: 
"[lpe] reinit x"
"[lpe] reinit x"
"[lpe] reinit x"
"[lpe] reinit x"
"[lpe] reinit x"
"[lpe] reinit x"
"[lpe] reinit x"
....
...
...
....
...

then at the same time, the data link is lost.

@LorenzMeier 
@jgoppert 
@mhkabir 
@sanderux

jgoppert, 2016-09-25:
This could suggest there is a nan being fed to lpe via a sensor. Do you have a log file?

JohnSnowball, 2016-09-25:
You mean that it's some sensor continuously sends NAN data to LPE?
If so, why the whole system is blocked? It looks like that only the LPE is running....
And as the system is blocked, sorry I don't have the log for the bug.
@jgoppert

jgoppert, 2016-09-25:
@JohnSnowball how can I try to reproduce this. Are you running in SITL?

JohnSnowball, 2016-09-25:
That's a long strory...A few weeks ago, I suffered a crush with a standard vtol:
http://discuss.px4.io/t/standard-vtol-lost-control-during-alt-hold-mode/1008/3

Then I made a matlab program to try to reproduce that bug.
And this time, I encountered the bug in the issue(no crash, just simulation).
The plane has something in common with last crush:
1. the whole system blocked
2. the data link is lost and can't regain by itself.
3. things happen during transition mc -> fw

jgoppert, 2016-09-25:
@JohnSnowball  I just tried vtol simulation in gazebo and transitioned to vtol and back, I can't reproduce the bug on master. What is the git hash of your software version?

build command:

```bash
make posix_sitl_default gazebo_standard_vtol
```

```bash
INFO  [simulator] Got initial simuation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
INFO  [tone_alarm] startup
WARN  [vtol_att_control] failed setting min values
WARN  [vtol_att_control] failed setting max values
WARN  [vtol_att_control] failed setting min values
.
.
INFO  [mavlink] mode: Normal, data rate: 2000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 2000000 B/s on udp port 14557 remote port 14540
INFO  [sdlog2] [blackbox] rootfs/fs/microsd/log/2016-09-25
WARN  [sdlog2] sdlog2: failed setting sched params
INFO  [sdlog2] [blackbox] recording: 07_27_05.px4log
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [local_position_estimator] [lpe] baro init 0 m std 0 cm
INFO  [local_position_estimator] [lpe] z resume 
INFO  [local_position_estimator] [lpe] gps init lat  47.40 lon   8.55 alt 488.1 m
INFO  [local_position_estimator] [lpe] xy resume 
INFO  [commander] home: 47.3977419, 8.5455939, 488.09
INFO  [tone_alarm] home_set

pxh> INFO  [tone_alarm] neutral
commander takeoff
pxh> INFO  [commander] Taking off
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977440, 8.5455936, 488.18
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected

pxh> commander tran[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/standard_vtol/gazebo/command/motor_speed, deleting message. This warning is printed only once.
sition
pxh> WARN  [vtol_att_control] failed setting max values
WARN  [vtol_att_control] failed setting min values

pxh> 
pxh> 
pxh> commander status
WARN  [commander] type: forward motion
WARN  [commander] safety: USB enabled: [NO], power state valid:  [OK]
WARN  [commander] avionics rail:  -1.00 V
WARN  [commander] home: lat = 47.3977440, lon = 8.5455936, alt = 488.18, yaw: 0.01
WARN  [commander] home: x = 0.2309987, y = -0.0242839, z = -0.10 
WARN  [commander] datalink: OK
WARN  [commander] main state: 10
WARN  [commander] nav state: 17
WARN  [commander] arming: ARMED
pxh> 
pxh> 
pxh> commander transition
pxh> WARN  [vtol_att_control] failed setting max values
WARN  [vtol_att_control] failed setting min values

pxh> 
pxh> 
pxh> commander status
WARN  [commander] type: symmetric motion
WARN  [commander] safety: USB enabled: [NO], power state valid:  [OK]
WARN  [commander] avionics rail:  -1.00 V
WARN  [commander] home: lat = 47.3977440, lon = 8.5455936, alt = 488.18, yaw: 0.01
WARN  [commander] home: x = 0.2309987, y = -0.0242839, z = -0.10 
WARN  [commander] datalink: OK
WARN  [commander] main state: 10
WARN  [commander] nav state: 17
WARN  [commander] arming: ARMED
```

JohnSnowball, 2016-09-25:
@jgoppert 
The firmware version is master a few weeks ago.
It's not SITL, the sensor data comes from the Matlab program. I guess it's the bad sensor data which caused the bug.
But if so, why the whole system is blocked and data link is lost?
At least the LPE is still runing during the bug, why data link is lost? I see no relation between bug an data link lost.

jgoppert, 2016-09-25:
The fastest that LPE will run is 100 Hz. I'm guessing that the accelerometer data is returning NaN. There is no check for this in the predict function. Then on the next step it detects the NaN in the state for LPE and re-initializes it. There is no reason that this would block the data link to my knowledge. Possibly your matlab program is not sending any valid data at that point and it sends a NaN somewhere else in the system that locks a more vital process.

jgoppert, 2016-09-25:
@JohnSnowball have you tried running ekf2 and seeing if you get the same error?

JohnSnowball, 2016-09-25:
@jgoppert  I reproduced the bug twice again! I hope you could take a look at the log file, especially for the accelerometer data as I can't confirm that was NAN data.
http://logs.uaventure.com/view/79xk2ELByoFru3wzyX4PMH
http://logs.uaventure.com/view/6purs8A3QRwjgWmaxN8NYM

If you think it's not NAN data,  I can make a NAN data manually and send it by matlab.
[breaklog.zip](https://github.com/PX4/Firmware/files/491541/breaklog.zip)

JohnSnowball, 2016-09-25:
btw, I never used ekf2, as I can't understand the code, it seems too different from a normal ekf predict->correct process.

jgoppert, 2016-09-25:
Can you also reproduce it on latest master?

JohnSnowball, 2016-09-25:
@jgoppert ok, but it will take a little bit time.

JohnSnowball, 2016-09-25:
@jgoppert 
I did another test, and got 2 more things make me confused....
I tried to use pos_estimator_inav to replace LPE to see if it works well.
1. SYS_MC_EST_GROUP, this parameter, I can't modify it in QGC, everytime I change it and restart px4, this value goes back to LPE. Finally, I modify this value through console.... Why I can't change that value with QGC? Shall I make some modification somewhere else?
2. The bug can't be reproduced with pos_estimator_inav, I guess it's LPE that requires better data quality, it can't handle bad data as well as inav. I wish you could give some comment.

And, if my plane doesn't have enough control ability, should I choose pos_estimator_inav instead of LPE?

jgoppert, 2016-09-25:
You should be able to set mc est group in qgc and it should stick after reboot. I saw a similar issue on master a few weeks ago. Please try updating to latest master.

JohnSnowball, 2016-09-27:
@jgoppert  hi, I found that bug is not in LPE, it's caused by other thread. I opened another issue, could you take some time to read that?
https://github.com/PX4/Firmware/issues/5553

closed 2016-09-27 by jgoppert