PX4/PX4-Autopilot #5547: SENS_BOARD_ROT other than 0 cause incorrent result

chobitsfan, 2016-09-26:
In 3dr solo SENS_BOARD_ROT = 12 (pitch 180)
It works OK in v1.3.4
but it does not work in v1.4.4 and v1.5(orientation is incorrect when display in QGC)

julianoes, 2016-09-27:
@bkueng did you maybe touch this in the sensors refactor?

LorenzMeier, 2016-09-27:
Code review says that the change was consistent and correct.

@chobitsfan Have you configured the vehicle fresh again and validated things are correct?

chobitsfan, 2016-09-28:
thank you :)

It happens after I build & upload master to solo
I set airframe to dji f450 to reset parameters, and set back to solo (incorrect orientation)
I use git checkout to v1.4.4, build & upload, reset all parameters (incorrect orientation)
then I git checkout to v1.3.4, build & upload, reset all parameters (everything ok)

I will do it again (tomorrow, after tornado pass)
I will flash it to v1.3.4, choose another airframe to reset parameters, then choose 3dr solo.
then I will flash it to v1.4.4, choose another airframe to reset parameters, then choose 3dr solo.

chobitsfan, 2016-09-28:
I have tested it again
I flash solo with stable (v1.4.1) and master, both show incorrect orientation in QGC
I check out v1.3.4, build & upload, correct orientation

limhyon, 2016-10-30:
Hi,
I have exactly same problem with @chobitsfan .
I flashed latest github master, and select vehicle configuration to 3DR solo.
It says, the IMU orientation is "Pitch 180" however, when I calibrate the accelerometer,
it show wrong orientation (for example, stand normal shows it is reversed).

limhyon, 2016-10-30:
I didn't isolate the problem whether this is becasue QGC problem or PX4 problem.
But ROTATION_YAW_270 in calibration menu gives correct result.

LorenzMeier, 2017-01-04:
Closing this as stale.

closed 2017-01-04 by LorenzMeier