PX4/PX4-Autopilot #5551: VTOL: erratic back transition

sanderux, 2016-09-27:
Back transitions usually produce erratic behavior.
Example: http://logs.uaventure.com/view/HfHt6Q2nbV5vKK3cYT6mYQ

Possible reasons for this:
- Velocity too high for MC when coming out of transition
- MC brake action results in pitch up, creating lift, and altitude increase. This in turn results in reducing thrust and therefor less stability
- MC overshoots it's position setpoint (in auto modes), the vtol tries to yaw to go back to the setpoint. Yaw is always hard on a vtol.
- A combination of the above

Possible solutions:
- Active breaking (like reverse pusher or flaps)
- Passive braking (like controlled stall / altitude gain)
- Weathervane (ignore yaw) until things return to normal
- Wait until the vtol has slowed down enough before attempting new position setpoints

Every solution comes with it's own problems and perhaps a combination of solutions is required.

@AndreasAntener  @dagar @tumbili @LorenzMeier

sanderux, 2016-09-27:
Reference: https://github.com/PX4/Firmware/issues/4845
Reference: https://github.com/PX4/Firmware/pull/5055

tubeme, 2016-10-01:
That is what we are experiencing all the time. Especially nasty it becomes when you flip the switch during turning the plane. Then the YAW is totally lost... It looks like it will crash, we had some crashes this way. 

The other way to brake the transition is when you are under power and make the back transition. Like 20-30% power on manual. That always goes in erratic behavior. It is like the quad does not have power and coordination to handle the plane.

This never happens with our KK20 benchmark system. KK always manages to recover and to handle the plane. With the KK system we are able to make front and back transition in pretty hard roll up to 20-30%. Which is not possible with the PX4 for now.

closed 2017-01-20 by sanderux