PX4/PX4-Autopilot #5553: sdlog and mavlink thread break down during transition

JohnSnowball, 2016-09-27:
I'm searching a bug which leads a crash, the phenomena is described as below:
standard vtol, during transition from mc to fw, right after the transition is finished(about 0.5sec), both QGC and RC link are lost, then crash...

I tried to reproduced that with the help of my matlab semi-physic simulation several times, for 10 times simulation, I will get one phenomena just as the crash.

To find out which part is down, I tried add "warnx(xxxxxx)" in each thread one by one as the code below:
https://github.com/CornerOfSkyline/TJ_Sim/commit/219032f05f551dd4ec8bedf6ef12bede98e2f950
(also all the code is in that branch)

Finally, I found that at the moment that bug occurs, it's the **sdlog2** and **mavlink** that stopped printing out, then, other thread like fw_att_control not working well, data link lost....

What kind of bug can cause phenomena like that, it's really fatal error....

mhkabir, 2016-09-27:
You're seem to be running out of stack.
Related issue here : https://github.com/PX4/Firmware/issues/5526

LorenzMeier, 2016-09-27:
@JohnSnowball Is this on hardware or in simulation?

JohnSnowball, 2016-09-27:
@mhkabir ok, I will check.
@LorenzMeier First time crash in a real flight, then afterwards in simulation. The phenomena are the same.

CornerOfSkyline, 2016-09-27:
It happened on hardware, we use matlab communication with the pixhawk through the serial 4 port. This is not happen everytime, happened one time in about 7-12 simulation. All happened after finish the translation from mc to fw. We check the log , about 0.5 second after translation finish.

CornerOfSkyline, 2016-09-28:
We can check the memory usage through the top command,but when the system down,the console have no respond but It can print warnx message.

LorenzMeier, 2016-09-28:
Can you share the complete log folder? Try to disarm when it is in that crashed state.

CornerOfSkyline, 2016-09-28:
@LorenzMeier I can share the complete log folder. I will take the test tommorw. But i think i cann't disarm the autopilot in that crashed state. It not responded the qgc and rc ,but i check the pwm output, it's repsonded the safety switch.

JohnSnowball, 2016-09-28:
@LorenzMeier       In that crash, we were unable to disarm as data link and rc both lost.

CornerOfSkyline, 2016-09-29:
[pix crashed sd file.zip](https://github.com/PX4/Firmware/files/500499/pix.crashed.sd.file.zip)

@LorenzMeier I tried today through the matlab simulator , in the thirteen time pixhawk crashed （the sess06 not corret simulate , the communiton was not right between pixhawk and matlab）.The sess13 is the crashed time log folder. The "pix crashed sd file.zip" is the whole sd file today ， i clean the sd card before to trace the bug.

CornerOfSkyline, 2016-09-29:
@LorenzMeier  @mhkabir @JohnSnowball 
I check the log file in sess1 preflight_perf001 and the postflight_perf001, i found that the fw_pos_ctrl_l1's stack almost full, so i try to decrease the fw_pos_ctrl_l1's stack size from 1400 to 1000, the system crashed sometime when vtol try to translate or translate finished. But it's different to the crashed before. In the crash before the console can print the warnx info, but i can not input the commander. In this session the console will print the hardfault info like this， and the system total dowm.
================================================================================
WARN  [commander] cm
WARN  [mavlink] mv
WARN  [mc_pos_control] mp
WARN  [mavlink] mv
WARN  [mavlink] mv
WARN  [sdlog2] s2
WARN  [mavlink] mv
WARN  [mavlink] mv
WARN  [mc_att_control] ma
WARN  [fw_att_control] fa
WARN  [vtol_att_control] vt
WARN  [mavlink] mv
WARN  [navigator] mp
WARN  [commander] cm
WARN  [mavlink] mv
WARN  [sdlog2] s2
WARN  [mavlink] mv
WARN  [fw_pos_control_l1] fl
WARN  [mc_pos_control] mp
WARN  [mc_att_control] ma
WARN  [mavlink] mv
WARN  [mavlink] mv
WARN  [fw_att_control] fa
WARN  [vtol_att_control] vt
WARN  [mavlink] mv
WARN  [commander] cm
WARN  [sdlog2] s2
WARN  [mavlink] mv
WARN  [mavlink] mv
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20005d78
IRQ stack:
  base: 20005dbc
  size: 000002e8
20005d60: 080d8460 000000b8 080d839a 20005d78 10001528 080a4a49 080d8460 000000b8
20005d80: 0809f089 00000010 200040c8 00000003 00000000 080a4cd9 080a4cc5 080ad1a1
20005da0: 00000000 00000000 100014c4 0001b898 e8ca5bf8 080a4c49 100013cc 00000000
sp:     100014a0
User stack:
  base: 10001528
  size: 0000063c
100014a0: 1000f190 00000000 0008c0eb 00000000 00000000 00000000 00000000 08032879
100014c0: 1000f190 0008c0eb 00000000 490c0eb0 1000f190 00000000 0008c0eb 080328f7
100014e0: 1000f190 00000005 20003f54 08032931 1000f190 20003f58 1000f194 080a264b
10001500: 080a267d 00000000 00000000 00000000 00000000 080a2685 080a267d 080a055b
10001520: 00000000 00000000 ffffffff 9c4b9be8 66406d0f d3e34b0f 000001a0 80000650
R0: 00000000 2002fff8 e8ca5b38 08cc7ae0 00000000 100014c4 0001b898 e8ca5bf8
R8: e8ca5b38 0000000d 20003f84 00000000 00000000 100014a0 0809fd91 080a2980
xPSR: 81000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9

LorenzMeier, 2016-09-29:
Can you share what you mean by almost full?

CornerOfSkyline, 2016-09-29:
[image]

In the picture , the left side is the preflight_perf and the right side is the postflight_perf. In the postflight_perf the fw_pos_ctrl_l1 stack is 1356/1392.

sanderux, 2016-09-29:
@CornerOfSkyline your last commit hash is unknown, did you change anything to the fw_pos_ctrl_l1 ?

LorenzMeier, 2016-09-29:
I just did a branch which increases stack sizes. Could you both try this one?
https://github.com/PX4/Firmware/pull/5569

@sanderux Sounds a bit like this is needed to have safe margins and it should go into master. But first I want to know if this fixes the systems.

sanderux, 2016-09-29:
@LorenzMeier i'm on an august branch and have no such problems. did this change recently?

LorenzMeier, 2016-09-29:
No. Stack overflows are not something deterministic. As soon as the stack is too small you can get lucky - or not. So the fact you didn't experience issues doesn't mean its ok for you.

sanderux, 2016-09-29:
hm, looking at a recent postflight it seems indeed a bit tight:
```
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  140292 35.923     0/    0   0 (  0)  READY 
   1 hpwork                      13233  2.650   804/ 1592 192 (192)  w:sig 
   2 lpwork                       1818  0.274   572/ 1592  50 ( 50)  READY 
   3 init                         1601  0.000  1168/ 2496 100 (100)  w:sem 
  93 gps                          1026  0.182   884/ 1192 220 (220)  w:sig 
  95 dataman                        64  0.274   700/ 1192  90 ( 90)  w:sem 
 122 sensors                     22515  4.661   916/ 1496 250 (250)  w:sem 
 124 commander                   10816  2.010  2692/ 2992 140 (140)  READY 
 126 commander_low_prio            633  0.000   804/ 2992  50 ( 50)  w:sem 
 135 px4io                       16474  3.473   948/ 1392 240 (240)  w:sem 
 141 mavlink_if0                  5525  1.188  1732/ 2792 100 (100)  READY 
 142 mavlink_rcv_if0                35  0.000   980/ 2096 175 (175)  w:sem 
 149 mavlink_if1                  4469  0.914  1708/ 2792 100 (100)  w:sig 
 150 mavlink_rcv_if1               274  0.000  1444/ 2096 175 (175)  w:sem 
 185 sdlog2                      16464  0.000  2492/ 3392 177 (177)  RUN   
 219 ekf2                        87753 16.544  5284/ 8496 250 (250)  w:sem 
 229 vtol_att_control            48817  9.872   684/ 1096 245 (245)  w:sem 
 232 mc_att_control              20263  4.296  1236/ 1496 250 (250)  w:sem 
 234 mc_pos_control              23392  2.833  1572/ 1896 250 (250)  w:sem 
 236 fw_att_control              17227  3.473  1116/ 1392 250 (250)  w:sem 
 241 fw_pos_ctrl_l1               3711  0.365  1356/ 1392 250 (250)  w:sem 
 247 navigator                   18663  2.193  1020/ 1296 105 (105)  READY 
```

CornerOfSkyline, 2016-09-29:
I think the reason is the fw_pos_ctrl_l1 stack overflow. I tried to change the fw_pos_ctrl_l1 to 1350, which is below the 1392 a bit and i try again. The pixhawk crashed when the transition finished, and the mavlink and sdlog thread down. The other thread is still alive but there is no hardfault info printed in the console.
@LorenzMeier @sanderux  @JohnSnowball

CornerOfSkyline, 2016-09-29:
@LorenzMeier  We tried the task stack in the #5569 twenty five times, it's ok. 
[sess031.zip](https://github.com/PX4/Firmware/files/500906/sess031.zip)

LorenzMeier, 2016-09-30:
I've merged the PR and close this now. Thank you for reporting and testing!

closed 2016-09-30 by LorenzMeier