PX4/PX4-Autopilot #5556: Build of posix_sitl_replay for EKF2 replay fails, attempts to write files to root fs

JGSOpenSrc, 2016-09-27:
Hi everyone,

So this is strange; I invoked make for the target posix_sitl_replay in order to do an EKF2 replay for tuning. The PX4 stack builds correctly and reaches [100%], but the make command fails because the recipe attempts to write two files to the root fs (and obviously fails due to permissions).

The files are replay_params.txt and replay.px4log

I think these files are supposed to be written in the same directory as the logfile parameter that is passed to make.

FYI I just did this with my local copy of the master branch, but I think the problem exists on 1.4.x releases.

Here's the console dump (its not terribly helpful but I suppose someone will ask for it).

> daniellab@ubuntu:~/src/smellocopter/firmware$ make posix_sitl_replay replay logfile=/home/daniellab/2016-09-09/07_10_20.px4log
> PX4 CONFIG: /home/daniellab/src/smellocopter/firmware/build_posix_sitl_replay
> [  0%] Built target git_genmsg
> [  0%] Built target prebuild_targets
> [  4%] Built target df_driver_framework
> [  5%] Built target git_gencpp
> [  6%] Built target git_mavlink
> [  6%] Built target git_uavcan
> [  6%] Built target git_matrix
> [  7%] Built target xml_gen
> [  7%] Generating uORB topic multi headers for posix
> [  8%] Built target git_ecl
> [  8%] Generating uORB topic headers
> [  8%] Generating uORB topic headers
> [  8%] Generating uORB topic multi headers for posix
> [ 47%] Built target msg_gen
> [ 48%] Built target platforms__common
> [ 50%] Built target platforms__posix__px4_layer
> [ 51%] Built target drivers__boards__sitl
> [ 55%] Built target drivers__device
> [ 62%] Built target platforms__posix__work_queue
> [ 63%] Built target systemcmds__param
> [ 64%] Built target systemcmds__ver
> [ 65%] Built target systemcmds__perf
> [ 68%] Built target modules__uORB
> [ 69%] Built target modules__param
> [ 70%] Built target modules__ekf2
> [ 71%] Built target modules__ekf2_replay
> [ 72%] Built target modules__sdlog2
> [ 74%] Built target lib__controllib
> [ 76%] Built target lib__mathlib
> [ 77%] Built target lib__conversion
> [ 77%] Built target lib__mathlib__math__filter
> [ 78%] Built target lib__external_lgpl
> [ 87%] Built target lib__ecl
> [ 88%] Built target lib__geo
> [ 96%] Built target modules__systemlib
> [ 97%] Built target lib__DriverFramework__framework
> [ 97%] Built target lib__geo_lookup
> [ 98%] Built target modules__logger
> [100%] Built target px4
> args: /home/daniellab/src/smellocopter/firmware/build_posix_sitl_replay/src/firmware/posix/px4 posix-configs/SITL/init/replay none replay none /home/daniellab/src/smellocopter/firmware /home/daniellab/src/smellocopter/firmware/build_posix_sitl_replay
> SITL ARGS
> sitl_bin: /home/daniellab/src/smellocopter/firmware/build_posix_sitl_replay/src/firmware/posix/px4
> rcS_dir: posix-configs/SITL/init/replay
> debugger: none
> program: replay
> model: none
> src_path: /home/daniellab/src/smellocopter/firmware
> build_path: /home/daniellab/src/smellocopter/firmware/build_posix_sitl_replay
> empty model, setting iris as default
> Replaying logfile: /home/daniellab/2016-09-09/07_10_20.px4log
> touch: cannot touch '/replay_params.txt': Permission denied
> src/firmware/posix/CMakeFiles/replay.dir/build.make:57: recipe for target 'src/firmware/posix/CMakeFiles/replay' failed
> make[4]: *** [src/firmware/posix/CMakeFiles/replay] Error 1
> CMakeFiles/Makefile2:6207: recipe for target 'src/firmware/posix/CMakeFiles/replay.dir/all' failed
> make[3]: *** [src/firmware/posix/CMakeFiles/replay.dir/all] Error 2
> CMakeFiles/Makefile2:6214: recipe for target 'src/firmware/posix/CMakeFiles/replay.dir/rule' failed
> make[2]: *** [src/firmware/posix/CMakeFiles/replay.dir/rule] Error 2
> Makefile:2392: recipe for target 'replay' failed
> make[1]: *** [replay] Error 2
> Makefile:160: recipe for target 'posix_sitl_replay' failed
> make: *** [posix_sitl_replay] Error 2

JGSOpenSrc, 2016-09-27:
It appears that the offending code is [here](https://github.com/PX4/Firmware/blob/master/Tools/sitl_run.sh#L106).

Shouldn't these files be written at least in /tmp? Surely not in root fs!

bkueng, 2016-09-29:
fixed in #5566

closed 2016-09-29 by LorenzMeier