PX4/PX4-Autopilot #5557: Cant build PX4 filmware on snapdragon

mahesha-path, 2016-09-28:
while running the command " make eagle_default " i faced these errors

anuj.gupta@CPU-291U:~/src/Firmware$ make eagle_default
PX4 CONFIG: /home/anuj.gupta/src/Firmware/build_posix_eagle_default
[ 1%] [ 1%] Built target git_gencpp
Built target git_genmsg
[ 4%] Built target df_driver_framework
[ 4%] Built target prebuild_targets
[ 4%] [ 5%] [ 6%] Built target git_mavlink
Built target git_uavcan
Built target git_matrix
[ 6%] Built target generate_px4muorb_stubs
[ 7%] [ 7%] Built target git_ecl
Built target git_jmavsim
[ 7%] Built target git_gazebo
[ 7%] Built target mixer_gen
[ 8%] Built target xml_gen
[ 38%] Built target msg_gen
[ 39%] Built target generate_topic_listener
[ 39%] Built target platforms__common
[ 39%] [ 39%] Built target drivers__rgbled
Built target drivers__pwm_out_sim
[ 40%] [ 40%] Built target drivers__boards__sitl
Built target drivers__blinkm
[ 42%] Built target drivers__device
[ 43%] Built target drivers__qshell__posix
[ 44%] [ 45%] Built target drivers__led
[ 45%] Built target systemcmds__param
Built target systemcmds__mixer
[ 46%] Built target systemcmds__ver
[ 46%] Built target systemcmds__topic_listener
[ 46%] Built target modules__attitude_estimator_q
[ 47%] Built target modules__position_estimator_inav
[ 52%] Built target modules__mavlink
[ 53%] [ 53%] Built target modules__mc_pos_control
Built target modules__ekf2
[ 54%] Built target modules__mc_att_control
[ 57%] Built target modules__local_position_estimator
[ 57%] Built target modules__param
[ 59%] [ 60%] Built target modules__sensors
Built target modules__systemlib__mixer
[ 63%] Built target modules__uORB
[ 64%] Built target modules__sdlog2
[ 64%] Built target modules__dataman
[ 65%] Built target modules__muorb__krait
[ 69%] Built target modules__commander
[ 70%] [ 75%] Built target modules__simulator
Built target modules__navigator
[ 76%] Built target lib__controllib
[ 77%] Built target lib__mathlib
[ 77%] [ 77%] [ 77%] Built target lib__mathlib__math__filter
Built target lib__conversion
Built target lib__geo
[ 77%] [ 78%] Built target lib__geo_lookup
Built target lib__runway_takeoff
[ 78%] Built target lib__terrain_estimation
[ 78%] [ 85%] Built target lib__tailsitter_recovery
Built target lib__ecl
[ 88%] Built target platforms__posix__px4_layer
[ 88%] Built target lib__DriverFramework__framework
[ 88%] Built target modules__logger
[ 94%] Built target modules__systemlib
[ 98%] Built target platforms__posix__work_queue
Linking CXX executable px4
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lpthread
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lrt
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lpthread
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lrt
collect2: error: ld returned 1 exit status
make[3]: *** [src/firmware/posix/px4] Error 1
make[2]: *** [src/firmware/posix/CMakeFiles/px4.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [posix_eagle_default] Error 2
thanks

julianoes, 2016-09-28:
Hi @mahesha-path 
What distribution are you using, and what version? Ubuntu?
Which instructions did you use? The ones in dev.px4.io?

mahesha-path, 2016-09-29:
I am using ubuntu 14.04
Yes i followed the dev.px4.io for installing the toolchain for linux and followed all steps  and after that when i tried to build the code i faced this errors

julianoes, 2016-09-29:
@mahesha-path Thanks for the info. I'm assuming that's on a 64bit system?
I'll try to reproduce this in a 14.04 VM.

mahesha-path, 2016-09-29:
Yes its a 64 bit system.......i tried to find the lpthread and lrt in path /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/lib/gcc/arm-linux-gnueabihf/4.9.3/ but the lib file was missing

Thanks

julianoes, 2016-09-29:
What is your `$HEXAGON_ARM_SYSROOT`?

julianoes, 2016-09-29:
> What is your `$HEXAGON_ARM_SYSROOT`?

Actually it builds without this, so disregard it.

mahesha-path, 2016-09-29:
Yes

mahesha-path, 2016-09-29:
Actually the lib file are not present in the standard path . I tried to debug the 1 error and found libpthread.so was in /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/libc/usr/lib/libpthread.a
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/libc/usr/lib/libpthread.so
so i copied these lib file and paste it in /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/lib/gcc/arm-linux-gnueabihf/4.9.3/    and now its not showing that error but some other error

Linking CXX executable px4
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find /lib/libpthread.so.0
/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/bin/ld: cannot find /usr/lib/libpthread_nonshared.a
collect2: error: ld returned 1 exit status

So may be the standard path is different , so is there any possibility to change the standard path where it is searching the lib files

julianoes, 2016-09-29:
I tried to reproduce your issue and setup a fresh Ubuntu 14.04 64 bit in a VirtualBox.
I followed the steps in the Devguide and `make eagle_default` works as expected.

I can only suggest to try to pull the latest updates in Firmware and cross_toolchain and run the Qualcomm install scripts again:

```
./installv3.sh
./qrlinux_sysroot.sh
```

anujkumargupta, 2016-09-30:
Actually i did the same but this time build posix eagle default works fine but build qurt eagle default giving error.
the log is 

Linking CXX executable px4
[100%] Built target px4
-- qurt-eagle-default
-- CMAKE_INSTALL_PREFIX: 
-- cmake module path: /home/anuj.gupta/src/Firmware/cmake
DF Drivers: mpu9250;bmp280;hmc5883;trone;isl29501
-- CMAKE_C_FLAGS:   -D__CUSTOM_FILE_IO__ -O0 -mv5 -G0
-- CMAKE_CXX_FLAGS:   -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -O0 -mv5 -G0
-- CMAKE_C_FLAGS:   -D__CUSTOM_FILE_IO__ -O0 -mv5 -G0
-- CMAKE_CXX_FLAGS:   -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -O0 -mv5 -G0
-- The CXX compiler identification is Clang 3.5.0
-- The C compiler identification is Clang 3.5.0
-- The ASM compiler identification is Clang
-- Found assembler: /home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/hexagon-clang
-- Check for working CXX compiler: /home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/hexagon-clang++
-- Check for working CXX compiler: /home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/hexagon-clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - failed
-- Check for working C compiler: /home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/hexagon-clang
-- Check for working C compiler: /home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/hexagon-clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- catkin DISABLED
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.6", minimum required is "2.7") 
Adding DF driver: mpu9250
Adding DF driver: bmp280
Adding DF driver: hmc5883
Adding DF driver: trone
Adding DF driver: isl29501
Generated generate_px4muorb_stubs target
module_libraries = drivers__device;modules__sensors;platforms__posix__drivers__df_mpu9250_wrapper;platforms__posix__drivers__df_bmp280_wrapper;platforms__posix__drivers__df_hmc5883_wrapper;platforms__posix__drivers__df_trone_wrapper;platforms__posix__drivers__df_isl29501_wrapper;systemcmds__param;modules__attitude_estimator_q;modules__position_estimator_inav;modules__local_position_estimator;modules__ekf2;modules__mc_att_control;modules__mc_pos_control;modules__param;modules__systemlib;modules__systemlib__mixer;modules__uORB;modules__commander;modules__land_detector;drivers__gps;drivers__pwm_out_rc_in;drivers__qshell__qurt;lib__controllib;lib__mathlib;lib__mathlib__math__filter;lib__geo;lib__ecl;lib__geo_lookup;lib__conversion;lib__terrain_estimation;lib__runway_takeoff;lib__tailsitter_recovery;lib__DriverFramework__framework;platforms__common;platforms__qurt__px4_layer;platforms__posix__work_queue;modules__muorb__adsp
target_libraries = 
df_driver_libs = df_mpu9250;df_bmp280;df_hmc5883;df_trone;df_isl29501
module_external_libraries = 
QURT_LIB_INCS = 
QURT_LIB_LINK_LIBS = drivers__device;modules__sensors;platforms__posix__drivers__df_mpu9250_wrapper;platforms__posix__drivers__df_bmp280_wrapper;platforms__posix__drivers__df_hmc5883_wrapper;platforms__posix__drivers__df_trone_wrapper;platforms__posix__drivers__df_isl29501_wrapper;systemcmds__param;modules__attitude_estimator_q;modules__position_estimator_inav;modules__local_position_estimator;modules__ekf2;modules__mc_att_control;modules__mc_pos_control;modules__param;modules__systemlib;modules__systemlib__mixer;modules__uORB;modules__commander;modules__land_detector;drivers__gps;drivers__pwm_out_rc_in;drivers__qshell__qurt;lib__controllib;lib__mathlib;lib__mathlib__math__filter;lib__geo;lib__ecl;lib__geo_lookup;lib__conversion;lib__terrain_estimation;lib__runway_takeoff;lib__tailsitter_recovery;lib__DriverFramework__framework;platforms__common;platforms__qurt__px4_layer;platforms__posix__work_queue;modules__muorb__adsp;df_mpu9250;df_bmp280;df_hmc5883;df_trone;df_isl29501;m
-- GIT_TAG = v1.5.0-202-g1eca977
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anuj.gupta/src/Firmware/build_qurt_eagle_default
PX4 CONFIG: /home/anuj.gupta/src/Firmware/build_qurt_eagle_default
Scanning dependencies of target git_driverframework
Scanning dependencies of target df_mpu9250
[  0%] Scanning dependencies of target git_gencpp
Scanning dependencies of target git_genmsg
[  0%] Generating git_init_src_lib_DriverFramework.stamp
[  0%] [  0%] Generating git_init_Tools_gencpp.stamp
Generating git_init_Tools_genmsg.stamp
Building CXX object src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/MPU9250.cpp.o
[  0%] Built target git_driverframework
[  0%] [  0%] Built target git_genmsg
Built target git_gencpp
Scanning dependencies of target df_bmp280
[  0%] Scanning dependencies of target df_hmc5883
Building CXX object src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/MPU9250_mag.cpp.o
[  0%] [  0%] Building CXX object src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o
Building CXX object src/lib/DriverFramework/drivers/hmc5883/CMakeFiles/df_hmc5883.dir/HMC5883.cpp.o
Linking CXX static library libdf_hmc5883.a
Linking CXX static library libdf_bmp280.a
Scanning dependencies of target df_trone
hexagon-ar: fatal: hexagon-ar: fatal: Numeric user ID too large
Numeric user ID too large
[  1%] make[3]: *** [src/lib/DriverFramework/drivers/bmp280/libdf_bmp280.a] Error 1
make[3]: *** [src/lib/DriverFramework/drivers/hmc5883/libdf_hmc5883.a] Error 1
[... cut ...]
make[2]: *** [src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/all] Error 2
Linking CXX static library libdf_trone.a
hexagon-ar: fatal: Numeric user ID too large
make[3]: *** [src/lib/DriverFramework/drivers/trone/libdf_trone.a] Error 1
make[2]: *** [src/lib/DriverFramework/drivers/trone/CMakeFiles/df_trone.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [qurt_eagle_default] Error 2
 i just want to know what is these two error

 Detecting CXX compile features
-- Detecting CXX compile features - failed
hexagon-ar: fatal: Numeric user ID too large
Thanks

julianoes, 2016-09-30:
I see this:

```
hexagon-ar: fatal: hexagon-ar: fatal: Numeric user ID too large
Numeric user ID too large
```

anujkumargupta, 2016-09-30:
So how to solve this??

julianoes, 2016-09-30:
I have never seen this problem before I'm afraid. Can you try on another machine or with a different user maybe?

What's the output of `id -u` or `echo $UID`?

anujkumargupta, 2016-09-30:
809502539

julianoes, 2016-09-30:
Well, usually for local users that number is around 100x. Can you try with a local user? Or can you try on another machine?

anujkumargupta, 2016-09-30:
Thanks
I am trying in some other machine while running the installv3.sh it throws error
Hexagon SDK not installed /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/tools/qaic/Ubuntu14/qaic

complete error log

anuj.gupta@GPTest-LaptopU:~/cross_toolchain$ ./installv3.sh 
HEXAGON_INSTALL_HOME [/home/anuj.gupta] 
Using the following for installation:
HEXAGON_SDK_ROOT=/home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0
HEXAGON_TOOLS_ROOT=/home/anuj.gupta/Qualcomm/HEXAGON_Tools/7.2.12/Tools
Hexagon SDK not installed /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/tools/qaic/Ubuntu14/qaic

Installing HEXAGON_SDK to /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0
This will take a long time.

Verifying required tools were installed...
Running 'make' in /home/anuj.gupta/Qualcomm/Hexagon_SDK/3.0/tools/qaic...
~/cross_toolchain ~/cross_toolchain
Qaic's Another IDL Compiler: version 1.0.37.1219545
Copyright (c) 2012 QUALCOMM Inc. All rights reserved.
~/cross_toolchain
Verifying setup is complete...

The Hexagon Tools 7.2.12 version was not installed.
Re-install the Hexagon SDK and select the Hexagon Tools install option
sh ./downloads/https://developer.qualcomm.com/download/hexagon/qualcomm_hexagon_sdk_lnx_3_0_eval.bin -i swing

anujkumargupta, 2016-09-30:
Thanks and dont think about last comment , that was solved , now i am trying to build that make file in different machine

anujkumargupta, 2016-09-30:
Hi i tried the sam thing on different machine but got same error "hexagon-ar:fatal: Numeric user ID too large"

again i tried to run makefile for6-7 times and got this error
Built target lib__runway_takeoff
[ 83%] Built target lib__tailsitter_recovery
[ 83%] Built target lib__DriverFramework__framework
[ 86%] Built target platforms__qurt__px4_layer
[ 91%] Built target platforms__posix__work_queue
[ 92%] Built target modules__muorb__adsp
[ 98%] Built target modules__systemlib
Linking CXX shared library libpx4.so
Fatal: cannot recognize the format of file `../../drivers/device/libdrivers__device.a'. object format or given target machine (hexagon-unknown-elf) is incompatible.
make[3]: *** [src/firmware/qurt/libpx4.so] Error 1
make[2]: *** [src/firmware/qurt/CMakeFiles/px4.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [qurt_eagle_default] Error 2

i tried to fine libpx4.so in my pc using locate libpx4.so command but there is no such shared library.

Thanks

julianoes, 2016-09-30:
What is the user ID on the different machin? Of course, it would have to be one where the user ID is just the usual 100x.

anujkumargupta, 2016-10-04:
Thanks @julianoes , that problem is silved using a virtual machine .
I buld the makefile and upload it in the Snapdragon FC but after that when i am trying it connect with QGC , QGC is not detecting it.
Do you have any idea....
Thanks

julianoes, 2016-10-04:
Thanks for the feedback. I'm closing this issue, please file a new one for the QGC connection issue.
Alse, there please provide the stdout and mini-dm output and some info about how the network is setup.

closed 2016-10-04 by julianoes

pyh0603, 2017-08-16:
Hi,dev,
          I compiled the latest PX4 version encountered some problems in Snapdragon flight, my git tag is EAGLE_DRONE_1.2_ES2_0. My VM is ubuntu 14.04 64-bit. I have build HelloWorld successfully.When i run make_eagle_leagcy_default the following problems:
mhhk@ubuntu:~/develop/Firmware$ make eagle_legacy_default 
-- posix-eagle-legacy
-- The ASM compiler identification is GNU
-- Found assembler: /home/mhhk/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/bin/arm-linux-gnueabihf-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
Generated generate_px4muorb_stubs target
-- GIT_TAG = EAGLE_DRONE_1.2_ES2_0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mhhk/develop/Firmware/build_posix_eagle_legacy_driver_default
PX4 CONFIG: posix_eagle_legacy_driver_default
Scanning dependencies of target git_genmsg
Scanning dependencies of target prebuild_targets
Scanning dependencies of target git_gencpp
[  0%] Generating git_init_Tools_genmsg.stamp
Scanning dependencies of target df_driver_framework
[  0%] Built target prebuild_targets
[  0%] Generating git_init_Tools_gencpp.stamp
Scanning dependencies of target generate_px4muorb_stubs
[  0%] Built target git_genmsg
[  0%] Generating px4muorb.h, px4muorb_skel.c, px4muorb_stub.c
Scanning dependencies of target git_uavcan
[  0%] Built target git_gencpp
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.obj
[  0%] Generating git_init_src_modules_uavcan_libuavcan.stamp
Scanning dependencies of target git_mavlink
[  0%] Built target git_uavcan
[  0%] Generating git_init_mavlink_include_mavlink_v1.0.stamp
Scanning dependencies of target git_matrix
[  0%] Built target git_mavlink
[  0%] Generating git_init_src_lib_matrix.stamp
Scanning dependencies of target xml_gen
[  0%] Built target git_matrix
[  0%] Generating parameters.xml
Scanning dependencies of target mixer_gen
[  1%] Generating mixer_multirotor.generated.h
……
……
[ 97%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/hx_stream.c.obj
[ 98%] Linking CXX static library libmodules__systemlib.a
[ 98%] Built target modules__systemlib
Scanning dependencies of target mainapp
[100%] Building C object src/firmware/posix/CMakeFiles/mainapp.dir/px4muorb_stub.c.obj
[100%] Building CXX object src/firmware/posix/CMakeFiles/mainapp.dir/__/__/platforms/posix/main.cpp.obj
make[3]: *** No rule to make target `/home/mhhk/Qualcomm/Hexagon_SDK/3.0/libs/common/rpcmem/UbuntuARM_Debug/rpcmem.a', needed by `src/firmware/posix/mainapp'.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [src/firmware/posix/CMakeFiles/mainapp.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [posix_eagle_legacy_driver_default] Error 2

I have follow the step as url:https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md.  but it still can not be solve.what should i do , hope for your reply! 
thank you very much!