PX4/PX4-Autopilot #5562: FOH alt mode does not work for transition waypoints

devbharat, 2016-09-28:
The gradient calculation for increment the altitude setpoint [here](https://github.com/PX4/Firmware/blob/92d160c431ab8e2721e95855f865748d9f5cc22f/src/modules/navigator/mission.cpp#L968-L977) after transition waypoint gets screwed because of [this function](https://github.com/PX4/Firmware/blob/92d160c431ab8e2721e95855f865748d9f5cc22f/src/modules/navigator/mission_block.cpp#L421-L432).

[image]

LorenzMeier, 2016-09-28:
Did you see that this has already been fixed on master?

dagar, 2016-09-28:
We could lazily initialize the previous altitude for FOH here instead of returning.
https://github.com/PX4/Firmware/blob/92d160c431ab8e2721e95855f865748d9f5cc22f/src/modules/navigator/mission.cpp#L920

devbharat, 2016-09-28:
Sorry about that https://github.com/PX4/Firmware/pull/5299 fixed it.

closed 2016-09-28 by devbharat