PX4/PX4-Autopilot #5573: SENS_EN_SF1XX missing

Shpiler, 2016-09-30:
Hi
I've tried all 1.4.x firmwares and i can't find SENS_EN_SF1XX parameter as said [here](http://dev.px4.io/sf1xx_lidar_setup.html). It doesn't appear even after full parameters reset.
[image]

Am i doing something wrong or it's just a problem with firmware itself? How can i enable my SF11/c LIDAR? 

QGC is 3.0.1

kd0aij, 2016-09-30:
Are you building for fmu-v4? It looks like that's the only config which
includes the sf1xx driver.

Shpiler, 2016-09-30:
> Are you building for fmu-v4?
I'm using pixhawk, so i have to use fmu-v2... or not?

kd0aij, 2016-09-30:
The sf1xx driver isn't listed here:
https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake

which probably explains why you can't find the parameter. You could try adding it here, but I don't know whether that will be sufficient, or whether it will fit in Pixhawk's flash.

Shpiler, 2016-09-30:
Thx, sound like I have to try unless we have another dev's word here.

Shpiler, 2016-10-03:
I added this driver to cmake file. The parameter has appeared, but lidar readings are still unavailable. The lidar itself works well via usb.

kd0aij, 2016-10-03:
That's progress. Is the driver being started? There should be something in the boot log and perhaps "ps" will show it.

ecmnet, 2016-10-04:
I had to remove some other drivers to get it into flash on px4fmu-v2. Enabling the sensor via parameter should then work out of the box. Btw. I connected my SF10a via I2C.

LorenzMeier, 2016-12-26:
Fixed on master in terms of startup and available on px4fmu-v3_default.

closed 2016-12-26 by LorenzMeier