PX4/PX4-Autopilot #5575: Move Sensor Covariance Parameters out of Estimators

jgoppert, 2016-09-30:
LPE/EKF2 currently has a ton of redundant parameters describing the sensor noise properties. I think it makes sense to move all noise properties within the uorb message and make the noise properties of sensors a global parameter in the SENS param namespace. This would mainly free up more memory for the more memory constrained boards.

kd0aij, 2016-09-30:
What is the bandwidth of the covariance estimate, and is there any consumer of that data besides the estimator which is currently active?

jgoppert, 2016-09-30:
I'm targeting all of these parameters:

https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/params.c

Single float values that describe the measurement noise (e.g. LPE_GPS_XY), not the entire covariance matrix.

LPE_GPS_XY describes the standard deviation of the GPS measurement that is assumed if eph isn't working. There is no reason that this is LPE specific and ekf2 has similar parameters.

kd0aij, 2016-09-30:
That *is* a lot of parameters. Are you proposing adding stdev to every sensor message, or should it be static for things like gyros and accelerometers?

jgoppert, 2016-09-30:
I think the easiest approach is just to move all of these parameters to the SENS namespace and then share between estimators, adding the parameters to uorb messages where it doesn't exist yet would probably be more work and might cause some more work for the cpu.

jgoppert, 2016-09-30:
The downside would be that all of the estimators would have to agree on the correct noise levels, which should only be sensor dependent, but we probably have some per estimator quirks currently.

jgoppert, 2017-01-28:
This is in progress on IEKF branch, @priseborough can you check if this looks reasonable for ekf2: https://github.com/PX4/Firmware/blob/d0d843750956ae7752b76e88d1af21df5676b69b/src/modules/iekf/params.c

priseborough, 2017-01-29:
There will be many estimator specific tuning requirements, even for those parameters that are common so I do not support sharing parameters between estimators.

jgoppert, 2017-01-29:
That was also my concern.  I thought it might work if we use these numbers as a noise floor and have estimator specific process noise added to them.

closed 2018-02-21 by PX4BuildBot