PX4/PX4-Autopilot #5576: Pixhawk won't boot when connected to PX4Flow

NCharabaruk, 2016-09-30:
Hello,

I have recently received a PX4Flow. It has been updated with the latest firmware. When it is connected to the Pixhawk, the Pixhawk won't boot. The LED blinks red and QGC says that the vehicle is not responding. The Pixhawk boots fine without the PX4Flow. 

I've made an extras.txt file in /fs/microsd/etc containing:
```
# Check for flow sensor
echo "waiting 4 seconds for PX4Flow to boot"
sleep 4
if px4flow start
then
  echo "PX4Flow start succeeded"
fi
```

This was recommended in another thread, but this changed nothing. Any help would be greatly appreciated.

kd0aij, 2016-09-30:
You may be having power issues if USB is the only source. Try booting on
battery power before connecting USB.

NCharabaruk, 2016-09-30:
Thanks, I had not thought of that. Unfortunately, it did not fix the problem. It still blinks red and can't connect.

kd0aij, 2016-09-30:
I'd connect a console terminal and look for messages there. You can find instructions on dev.px4.io.

jgoppert, 2016-09-30:
Do you see the lights on the px4flow start up the same as when they are powered via usb? If not, you might have the polarity on the i2c reversed.

jgoppert, 2016-09-30:
Also, what else is on your i2c bus? If you have a lot, you may be getting a brown out as @kd0aij suggested.

NCharabaruk, 2016-09-30:
The lights start the same powered by USB or battery. The green on light turns on and steady while the red com light flashes rapidly. After a few seconds the red still flashes but slower, and the blue act button pulses.

I have a compass, MB1222 sonar sensor, and the PX4Flow hooked up to the I2C bus.

jgoppert, 2016-09-30:
Yeah, if you try to power all of that on usb power it will brown out. Try battery power.

NCharabaruk, 2016-09-30:
I did try on battery power. Same result

jgoppert, 2016-09-30:
try unplugging some of the other i2c devices

NCharabaruk, 2016-09-30:
I've tried it without the sonar. Trying without the compass results in no change

mhkabir, 2016-10-01:
Remove the extras.txt file. The sensor is auto-started by default from the standard startup script.
Do you have a serial-USB adapter (e.g FTDI) which you can use to check the system boot console?

NCharabaruk, 2016-10-03:
I've taken the extras.txt file out again. I have ordered the adaptor but haven't received it yet.

mhkabir, 2016-10-03:
Ok. Do you know what the bootloader version on the flow board is?

NCharabaruk, 2016-10-03:
I don't know how to check the bootloader version. I just got the board last week and I update the firmware using QGC if that info helps.

NCharabaruk, 2016-10-05:
So I figured it out. For some reason, after plugging in the PX4Flow the Pixhawk could no longer detect the sonar sensor that was already attached to it. On booting, the bootloader never got past looking for the sonar sensor. Disabling it in the settings got the bootloader past this.

closed 2016-10-05 by NCharabaruk