PX4/PX4-Autopilot #5581: Adding Custom Configuration in sitl_gazebo

mohit004, 2016-10-01:
Dear Developers,

I am trying to add a new configuration (tilt-rotor tricopter) in the software-in-the-loop package of PX4 Firmware. In the PX4 developer's guide it is mentioned that in order to customize the gazebo simulation interface, one just has to modify the codes in 'Tool/sitl_gazebo' folder. However, while reading all the plugin codes in folder 'Tool/sitl_gazebo/src', I got confused as in the code itself I can see that the plugins are publishing on some ROS topics, but when I open 'rqt_graph', I am not able to see any connection between ROS node and GAZEBO node. In order to proceed, I would like to clarify few things:

1. Are the changes only required in 'Tool/sitl_gazebo' folder?
2. After I modify all the plugins, do I need to make any change in the control codes of Pixhawk's firmware, as for my configuration I have 3 motors and 1 servo? My intuition says that I dont need to, as the PX4 flightstack is configuration independent.
3. Any information or reference that I can read to follow.

I really appreciate any of your help!

Thankyou
Mohit

julianoes, 2016-10-11:
@mohit004 I don't know much about the subject, I'm sorry. However, you might be able to compare the iris and the tailsitter to see what's implemented where.

mohit004, 2016-10-11:
Dear @julianoes,

Thankyou for your reply. I have been looking at the codes and right now, trying to create the plugin for servo motor. I have some specific but quick doubts. So, if you could direct me to the concerned person, I would really appreciate it.

Thanks again for your help.

Kind Regards
Mohit

mohit004, 2016-10-18:
Dear @julianoes,

Thanks for giving @bkueng contact.

Dear @bkueng,
I just have a quick question regarding the servo plugin that I am trying to write.
In 'gazebo_motor_model' plugin of PX4 Firmware, the developer is using a 'FirstOrderFilter' object in order to obtain the rotational velocity at next time. However, in my case of a servo motor, I actually need a second order filter to map input voltage to the angular position. Is it Correct?
Also, about modeling the DC servo motor, do I really need to write the second order filter or can I control the servo motor by angular velocity (instead of angular position)?

Any of your help is highly appreciated.

Kind Regards
Mohit

bkueng, 2016-10-18:
Hi @mohit004 
I'm sorry, I don't know much about the subject as I did not write these plugins. But the documentation about mixing might help with your second question: http://dev.px4.io/concept-mixing.html.

mohit004, 2016-10-19:
Hi @bkueng,
Thankyou for your prompt reply.
Could you please give me the contact of the person who was involved in writing these plugins.
About mixing I have no problem as I was able to include my custom configurations on the Pixhawk hardware already. However, for this SITL I only saw one mixer file which to me does not explain the working of all the configurations that are available in the SITL simulation part of Pixhawk firmware.
Once again, thankyou for your help.
Mohit

mhkabir, 2016-10-19:
Don't we already have a servo model for the fixed wing and delta quad wing actuators?

mohit004, 2016-10-19:
Hi @mhkabir,
No, sorry I couldnot find any plugin that includes a servo model. In the 'liftdragplugin' they are directly providing a parameter 'control_joint_rad_to_cl', which to my understanding directly maps the control surface deflection (in radians) to the change in lift force.
Thanks
Mohit

mohit004, 2016-11-07:
Dear Developers,

Just an update about the work that I was doing. 
I have written the necessary plugins for tail motor and servo motor. In the servo plugin, all I am doing is to control the servo (joint) position using 'SetPosition' function of the gazebo joint class.
However, there are still few problems that I am facing.

**1. Problem in Servo Plugin.**
In the motor plugins that came with the SITL package, the joints are controlled using 'SetVelocity' function, which makes sense in case of a motor. But in case of controlling the angular rotation of a joint, i.e. in my case, controlling the tail servo position, doing it using 'SetVelocity' would be very difficult as in that case I have to find out the mapping from the controller output (coming from PX4) to the servo velocity, which is very difficult to find. Right now, what I did is that I found the mapping from PX4's controller output to the angular rotation of the servo. So in order to use this approach, I tried using 'SetPosition' function of gazebo, but I dont know something happens when I do that. To me it appears that the 'SetPosition' function of gazebo does not work properly, as it makes the joint stick in space. I am not sure about this point. I have also read on the internet that people have also faced some problems using 'SetPosition' function.

**2. Problem in controlling the tricopter itself.**
On account of problem 1, I chose an alternate way of not making the joint move at all, but passing the PX4 controller's output directly to the tail motor in order to apply the forces accordingly. When I display these forces, they appear to be fine. But, when I arm the tricopter and try to command the takeoff, it just shows some weird behavior, which I am not able to figure out. 

I have been trying to make all this work for months now but without a success. Dear @dagar, @julianoes, @bkueng, @mhkabir, @LorenzMeier, any of your kind help is deeply appreciated.

Kind Regards
Mohit

closed 2016-12-28 by LorenzMeier

ganeshrk95, 2017-05-11:
@mohit004 Were you able to get the tricopter simulation working ? I'm working on a similar model. Can't get the servo and back motor to arm/work together.

pyromania99, 2024-06-28:
@mohit004 I would also like to develop a servo plugin. Could you explain the process for it? Or share the plugin so I may modify it for my requirements