PX4/PX4-Autopilot #5586: Critical Issue mission arming

jcooper21, 2016-10-02:
I just loaded todays master and tried to do a mission as it turns out I cannot fly a mission with a plane.  When I select mission mode it says I have to arm in a manual mode then switch to mission mode.  When I do that the motor spins up on the ground and I cannot launch the plane by hand.  What happened to shake on for missions?

LorenzMeier, 2016-10-02:
It sounds like your radio control is not properly calibrated. Regarding the mission handling you're describing: It really needs someone who picks it up.

jcooper21, 2016-10-02:
@LorenzMeier The radio has been re calibrated many times so that should work, especially since it arms in stab and doesn't spin up the motor with zero throttle.  I have tested master and stable but neither of them seem to work, I have been flying 1.3 regularly without issues.  I have multiple parameter files that I have tried as well which all previously worked and so far nothing works.

LorenzMeier, 2016-12-28:
@jcooper21 Can you re-test the latest release?

dagar, 2017-01-14:
@jcooper21 this is a mission with a takeoff waypoint?
You want the motor stopped until you throw?
If so, have you looked at launch detection? https://pixhawk.org/firmware/parameters#launch_detection

dagar, 2017-04-11:
@jcooper21 is this still a problem?

dagar, 2017-04-28:
I believe this has been resolved. Please reopen if I'm wrong.

closed 2017-04-28 by dagar