PX4/PX4-Autopilot #5594: Missing __EXPORT in px4fmu_can.c ?

mainini, 2016-10-04:
Hi all

I noticed that src/drivers/boards/px4fmu-v2/px4fmu_can.c in the current stable tree is missing an __EXPORT for int can_devinit(void). This is the case also for the px4fmu-v1 and px4fmu-v4 boards.

Is this on purpose? It leads to trouble compiling CAN-examples...

Thanks for a comment on this issue.

davids5, 2016-10-04:
PX4 uses UAVCAN that provides it's own stm32 CAN driver, therefore the startup code does not init Nuttx's CAN driver.

LorenzMeier, 2016-10-06:
We should still fix it for the NuttX CAN examples I guess?

mainini, 2016-10-06:
That would be great, I think there are also usecases which use the plain CAN driver (I do).
I found it confusing that it gave me a compile error and I could not tell if this was some design decision or if I was doing something wrong...

Thanks!

davids5, 2016-10-06:
Ok I will ensure it is linkable when the NuttX example is enabled.

davids5, 2016-10-06:
@LorenzMeier - Done in https://github.com/PX4/Firmware/pull/5613

closed 2016-10-06 by davids5