PX4/PX4-Autopilot #5595: Airspeed temeperature dependency

AndreasAntener, 2016-10-04:
@LorenzMeier first shot at logging temperature data. Looks like I need to allow the negative values for the test. But I was wondering why one sensor gives a higher reading while the other one goes down? Both sensors were calibrated inside and covered. Also they show a different temperature. Are we sure that the voltage correction that happens in the driver is correct (both setups have a slightly different system voltage)? Maybe I need a different procedure to measure and log the raw values, maybe with generated airspeed.

pixfalcon-auva-outside
[image]

pixfalcon-auva-inside
[image]

dropix-3dr-outside
[image]

dropix-3dr-inside
[image]

Ref: #4616

sanderux, 2016-10-04:
Here's what happens when the sensor is mounted behind the air intake on a vtol:
[image]
http://logs.uaventure.com/view/jg2rHZDRmAboYLiqyFVuge

Here's what happens after a few flights:
[image]
http://logs.uaventure.com/view/DwRTKb7jKf5LHXH553s6n7

Here's what happens in africa with bad cooling and wrong angle:
[image]
http://logs.uaventure.com/view/5MFXSfPmYwrUFxuCDnQj7T

AndreasAntener, 2016-10-05:
@LorenzMeier @priseborough I'd appreciate feedback on this. It is possible that my measuring method makes no sense for what we want to find out (temperature dependency and possible method to compensate, or at least enforce a recalibration if needed) and that we should actually somehow generate constant airspeed for the test than just recording the offset.

Here's what I got: the following charts show the differential pressure (in Pa) and temperature as they are returned by the sensor and in addition the 5v corrected values (the correction that happens in the driver). I tested with the default 3DR (calibration offset ~6Pa) sensor and with the new one from AUAV (calibration offset ~60Pa).

The pitot tubes were lightly covered on both setups. The sensors are outside the body. Both setups were calibrated inside powered with the main battery. Inside temperature guessed around 25 degree, outside temperature roughly 10 degree.

**Test 1: cool down outside**

Dropix with 3DR sensor:
[image]

Pixfalcon with AUAV sensor:
[image]

**Test 2: start logging inside, only a brief moment outside**

Legend:

1. opening hallway door (cold air starts to flow in)
2. carrying vehicle outside
3. placing on the ground outside
4. carrying vehicle inside
5. placing vehicle on the ground inside
6. closing hallway door

Dropix with 3DR sensor:
[image]

Pixfalcon with AUAV sensor:
[image]

**Observations**
- both setups react into opposite directions with the temperature change
- the change is very noticeable going outside but not really when coming back in
- the 3DR sensor slowly changes with temperature where as the AUAV sensor shows change in a much smaller timeframe

This makes me believe that the 2 setups must have a difference I'm not aware of and that I need a different test setup/method :)

LorenzMeier, 2016-10-05:
Awesome data! Its possible that both sensors respond differently to temperature. @tumbili Could you maybe fit a polynomial and try compensating in the driver? I would not be surprised if you get a correct fit for both if you calibrate both that way.

tubeme, 2017-05-26:
Any progress in this research? We continue to have nasty problems with the airspeed related to temperature. There are cases where we get a 15m/s difference between morning and afternoon flights in California where the temp difference is huge between morning and afternoon. Everybody is waiting for the new 55 chip which states that has temp compensation, and indeed it has a totally different way of compensation than the old 45 chip. But sadly there is no delivery yet of this new chip, and nobody says when it is going to be available. For all the current 45 chips it is totally unreliable. And as I remember a discussion with @Lorenz one or two years ago this is probably the most important reading for the autopilot. Especially for the VTOL transition.  So what we can do in this situation?? Is there a mathematical way to compensate somehow this differences?

We got hold of couple of airspeed sensors from real small planes. On some chips they just use heater to give the sensor a constant temperature. On others they don't.

sanderux, 2017-05-27:
@tubeme https://docs.px4.io/en/advanced_config/vtol_without_airspeed_sensor.html

tubeme, 2017-05-27:
Thanks @sanderux We are preparing a machine for this use allready and we will try it. Are the TECS going to work in this scenario and prevent stalls during long mission flights? Is there a way to detect stall without the airspeed and trigger some reaction during flight in guided modes?

LorenzMeier, 2017-05-27:
We will test next week a sensor from SENSIRION. That should also help.

sanderux, 2017-05-27:
@tubeme i have found that tecs responds quite well when you have good min/max pitch angles set.

tubeme, 2017-05-27:
@LorenzMeier Which one? https://www.soselectronic.com/products/freescale/mpx-5500-dp-9006. Do you have a PCB for the test? We might buy some samples as well. Is there a driver for the PX4 or are you going to bench test it only in a lab? Thanks.

tubeme, 2017-05-27:
@sanderux Thanks for the info, we will find out in real conditions. :) Hope the plane will survive...

tubeme, 2017-06-02:
I have the MS5525 sensor for tests and it does not run under the current master. Is there anything that should be preset somewhere in the firmware in order for the FW to recognize it? Probably it need different driver?

dagar, 2017-06-27:
Closing this as I think we're close to moving on from the ms4525.

closed 2017-06-27 by dagar

njoubert, 2017-07-11:
@tubeme I know this issue is closed, but you linked to a MPX5500 sensor. I'm looking at that sensor, but I don't think that can make a useful airspeed sensor. with a pressure range from 0 to 500kPa, it's got two orders of magnitude higher maximum pressure range than any of the other sensors we use for airspeed on PX4. Did you link to the wrong sensor?