PX4/PX4-Autopilot #5596: Home point not set due to recent commit

bartslinger, 2016-10-04:
On stable, the drone sets the home point as soon as it gets a good gps position. This works in HITL (with xplane) and in the real world. On master, the home point does not get set. After disabling the changes made by https://github.com/PX4/Firmware/commit/50b8ed0a89d2a6022cc518df65fa6bc57375f393 , it does set the home point as before.

bartslinger, 2016-10-04:
I'll try although I am a bit scared of the commander.cpp monster

julianoes, 2016-10-05:
> I'll try although I am a bit scared of the commander.cpp monster

Hehe, but you've already narrowed it down a lot by finding the change that causes it :smile:.

bartslinger, 2016-10-07:
Turned out to be something else, originating from QGC. It sometimes sent out NaN on acceleration measurements.

closed 2016-10-07 by bartslinger