PX4/PX4-Autopilot #5597: Send custom MAVLink message

pietrop88, 2016-10-04:
Hi,

I want to send a custom uOrb message through mavlink. 
I follow the guide: http://dev.px4.io/custom-mavlink-message.html.
The class that I added to `mavlink_messages.cpp` is this one:
```
class MavlinkStreamCaTrajectory : public MavlinkStream
{
public:
    const char *get_name() const
    {
        return MavlinkStreamCaTrajectory::get_name_static();
    }
    static const char *get_name_static()
    {
        return "CA_TRAJECTORY";
    }
    static uint8_t get_id_static()
    {
        return MAVLINK_MSG_ID_CA_TRAJECTORY;
    }
		uint8_t get_id()
		{
			return get_id_static();
		}
    static MavlinkStream *new_instance(Mavlink *mavlink)
    {
        return new MavlinkStreamCaTrajectory(mavlink);
    }
    unsigned get_size()
    {
        return MAVLINK_MSG_ID_CA_TRAJECTORY_LEN + MAVLINK_NUM_NON_PAYLOAD_BYTES;
    }

private:
    MavlinkOrbSubscription *_sub;
    uint64_t _ca_traj_time;

    /* do not allow top copying this class */
    MavlinkStreamCaTrajectory(MavlinkStreamCaTrajectory &);
    MavlinkStreamCaTrajectory& operator = (const MavlinkStreamCaTrajectory &);

protected:
    explicit MavlinkStreamCaTrajectory(Mavlink *mavlink) : MavlinkStream(mavlink),
        _sub(_mavlink->add_orb_subscription(ORB_ID(ca_trajectory))),  // make sure you enter the name of your uorb topic here
        _ca_traj_time(0)
    {}

    void send(const hrt_abstime t)
    {
      struct ca_traj_struct_s _ca_trajectory;    //make sure ca_traj_struct_s is the definition of your uorb topic

      if (_sub->update(&_ca_traj_time, &_ca_trajectory)) {
        mavlink_ca_trajectory_t _msg_ca_trajectory;  //make sure mavlink_ca_trajectory_t is the definition of your custom mavlink message

        _msg_ca_trajectory.timestamp = _ca_trajectory.timestamp;
        _msg_ca_trajectory.time_start_usec = _ca_trajectory.time_start_usec;
        _msg_ca_trajectory.time_stop_usec  = _ca_trajectory.time_stop_usec;
        _msg_ca_trajectory.coefficients =_ca_trajectory.coefficients;
        _msg_ca_trajectory.seq_id = _ca_trajectory.seq_id;
        mavlink_msg_ca_trajectory_send_struct(_mavlink->get_channel(), &_msg_ca_trajectory);
      }
    }
};
```
I appended the stream class
```
StreamListItem *streams_list[] = {
...
  new StreamListItem(&MavlinkStreamCaTrajectory::new_instance, &MavlinkStreamCaTrajectory::get_name_static,&MavlinkStreamCaTrajectory::get_id_static),nullptr
};
```
Differences respect to the guide:

1. Since the constructor of class `StreamListItem` requires the id i added the method  `get_name_static` to `MavlinkStreamCaTrajectory` class.
2. In the send method of class `MavlinkStreamCaTrajectory` I changed the method
`_mavlink->send_message(MAVLINK_MSG_ID_CA_TRAJECTORY, &_msg_ca_trajectory);`
with:
`mavlink_msg_ca_trajectory_send_struct(_mavlink->get_channel(), &_msg_ca_trajectory);`
because I had a compile error.
Now I am able to compile the code but the function send is never used even if I send `ca_trajectory` uOrb message and I don't receive any `ca_trajectory` mavlink message.

I think that something is missed but I don't know what.

bkueng, 2016-10-05:
Try to enable the stream, like in https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/lpe/iris#L67

pietrop88, 2016-10-05:
Thanks for your reply. 
Now the function `send` in the `MavlinkStreamCaTrajectory` class is used and I suppose that the mavlink package is sent but I am not able to see it.
I'm using mavros to stream mavlink messages to the pc and if I check the topic `mavros/from` (http://wiki.ros.org/mavros#Subscribed_Topics), I cannot see my custom mavlink message.
I also checked the Mavlink inspector inside qgroundcontrol (that is connected through mavros) but there is not my message.

Do I miss anything?
Is there a better way to check the incoming Mavlink message?

bkueng, 2016-10-05:
You need to make sure QGC or mavros support your custom message as well. [pymavlink](http://qgroundcontrol.org/mavlink/pymavlink) might be a simple way to test your message.

pietrop88, 2016-10-06:
I extended mavros and now it works.
Thank you.

bkueng, 2016-10-10:
You're welcome

closed 2016-10-10 by bkueng

xj19941015, 2017-05-29:
i add a new mavlink messages ,but my qgc cannot receive it  .what is wrong