PX4/PX4-Autopilot #5600: SITL build on OSX with latest XCode (SDK 10.12) fails

DonLakeFlyer, 2016-10-04:
```
Donalds-Mac-Pro:Firmware Don$ make posix_sitl_default jmavsim -j16
PX4 CONFIG: /Users/Don/repos/Firmware/build_posix_sitl_default
make[1]: warning: -jN forced in submake: disabling jobserver mode.
[  1%] Built target prebuild_targets
[  2%] Built target git_matrix
[  2%] Built target git_genmsg
[  2%] Built target git_gencpp
[  2%] Built target git_jmavsim
[  2%] Built target git_mavlink
[  2%] Built target git_uavcan
[  2%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o
[  2%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DFDiag.cpp.o
[  2%] Built target git_ecl
[  2%] Built target git_driverframework
[  3%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/I2CDevObj.cpp.o
/Users/Don/repos/Firmware/src/lib/DriverFramework/framework/src/Time.cpp:44:9: fatal error: 
      'CLOCK_REALTIME' macro redefined [-Wmacro-redefined]
#define CLOCK_REALTIME 0
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:154:9: note: 
      previous definition is here
#define CLOCK_REALTIME _CLOCK_REALTIME
        ^
1 error generated.
make[4]: *** [src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o] Error 1
make[4]: *** Waiting for unfinished jobs....
[  4%] Built target xml_gen
[  4%] Built target git_gazebo
[  4%] Built target mixer_gen
make[3]: *** [src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 23%] Built target msg_gen
make[2]: *** [src/firmware/posix/CMakeFiles/jmavsim.dir/rule] Error 2
make[1]: *** [jmavsim] Error 2
make: *** [posix_sitl_default] Error 2
```

eric1221bday, 2016-10-04:
Hmm , i've just ran that exact command on my computer, which is also running Mac 10.12 SDK, and in fact has that exact line which generated your error in my SDK, but my build succeeded

DonLakeFlyer, 2016-10-04:
I tried all variants al make clean I could come up with since I thought maybe there was something old hanging around.

julianoes, 2016-10-05:
@DonLakeFlyer I'll try to replicate this tomorrow.

julianoes, 2016-10-10:
@DonLakeFlyer I can't reproduce this, however, I'm on 10.11.6, however with XCode 8.0.

@LorenzMeier any idea?

DonLakeFlyer, 2016-10-10:
Either some sort of clean problem or tool update I missed. Let me try a fresh repo and see what happens.

DonLakeFlyer, 2016-10-10:
I'm on XCode 8 as well. Tried a fresh repo and got this:
```
[  3%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/I2CDevObj.cpp.o
/Users/Don/repos/fw2/src/lib/DriverFramework/framework/src/Time.cpp:44:12: fatal error: 
      unused function 'clock_gettime' [-Wunused-function]
static int clock_gettime(int clk_id, struct timespec *t)
           ^
1 error generated.
make[4]: *** [src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framewo
```

LorenzMeier, 2016-10-10:
xcodeselect --install

DonLakeFlyer, 2016-10-10:
Yeah, that was it.

closed 2016-10-10 by DonLakeFlyer

nacho-carnicero, 2017-02-02:
Hi I'm getting similar problems when trying to compile with `posix_sitl_default` as target. I have the Command Line Tools installed (as it can be seen from the error message), but I still get the following error message:
```
make posix_sitl_default
dyld: Library not loaded: /usr/local/opt/readline/lib/libreadline.6.dylib
  Referenced from: /usr/local/bin/awk
  Reason: image not found
-- PX4 VERSION: v1.6.0rc1-84-g3dc6e7b
-- CONFIG: posix-sitl-default
-- CMAKE_INSTALL_PREFIX: /usr
-- CMAKE_MODULE_PATH: /Users/nachocarnicero/src/Firmware/cmake
-- The CXX compiler identification is AppleClang 8.0.0.8000042
-- The C compiler identification is AppleClang 8.0.0.8000042
-- The ASM compiler identification is AppleClang
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: //anaconda/bin/python (found version "2.7.11") 
-- C compiler: Apple LLVM version 8.0.0 (clang-800.0.42.1)
-- C++ compiler: Apple LLVM version 8.0.0 (clang-800.0.42.1)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/nachocarnicero/src/Firmware/build_posix_sitl_default
[24/495] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o
FAILED: /Library/Developer/CommandLineTools/usr/bin/c++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_SITL -D__DF_DARWIN -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I. -Isrc -Isrc/modules -Isrc/modules/px4_messages -I../mavlink/include/mavlink -I../src -I../src/drivers/boards/sitl -I../src/include -I../src/lib -I../src/lib/DriverFramework/framework/include -I../src/modules -I../src/platforms -I../src/lib/matrix -I../src/modules/systemlib -I../src/platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -Wall -Werror -Wextra -Wno-sign-compare -Wshadow -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=format-security -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=reorder -Werror=uninitialized -Werror=init-self -Qunused-arguments -Wno-unused-const-variable -Wno-varargs -Wno-address-of-packed-member -Wno-unknown-warning-option -Wno-missing-field-initializers -fvisibility=hidden -include visibility.h -O2 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fcolor-diagnostics   -include visibility.h -std=gnu++11 -MD -MT src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o -MF src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o.d -o src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o -c ../src/lib/DriverFramework/framework/src/Time.cpp
../src/lib/DriverFramework/framework/src/Time.cpp:44:9: fatal error: 'CLOCK_REALTIME' macro redefined [-Wmacro-redefined]
#define CLOCK_REALTIME 0
        ^
/usr/include/time.h:154:9: note: previous definition is here
#define CLOCK_REALTIME _CLOCK_REALTIME
        ^
1 error generated.
[24/495] Generating parameters.xml
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1
```

I'm in Mac Os Sierra (10.12.2) with Xcode 8.2.1

Any help would be appreciated!

Cheers, 

Nacho.

LorenzMeier, 2017-02-02:
You're missing parts of the command line tools / support libraries. You don't have the POSIX headers. Once you install them you should be fine.

nacho-carnicero, 2017-02-03:
Thanks for the quick reply! Indeed, I had to update `gawk` that was linked to an old version of the `readline` library.
A `brew update gawk` solved this issue.

Thanks again!