PX4/PX4-Autopilot #5604: INAV estimated position cant reset after arming

meteor0530, 2016-10-05:
Hi,

My system is OSX, I recently updated the PX4 firmware through QGC 3.0.0. I used INAV estimator and GPS for position estimate, then I used mavlink inspector to monitor the local_position_ned, and found after I armed the FCU the position didn't reset (x,y,z cant become 0). how can I fix it?

Any help is appreciated.

julianoes, 2016-10-05:
Hi, I don't think the local position resets on arming. It only "resets" when the estimator initializes.

What does reset on arming however, is the home position.

meteor0530, 2016-10-05:
Thanks, maybe my description was unclear. when I used the pervious version of the firmware, the x,y,z coodinates in LOCAL_POSITON_NED reset to 0 after I armed. After I updated the firmware, it didin't work.
And I  don't understand the meaning of the home position, which kind of position does the LOCAL_POSITION_NED belong to?

julianoes, 2016-10-05:
Ok, I might be wrong. @tumbili can you help?

tumbili, 2016-10-05:
@meteor0530 @julianoes INAV sets it's reference origin [here](https://github.com/PX4/Firmware/blob/master/src/modules/position_estimator_inav/position_estimator_inav_main.cpp#L869) and I don't see any dependency on arming there. Basically the origin is set as soon as the gps quality is sufficient.
When it does that you should get this message in QGC

[inav] init ref: ...

Are you sure that you are using INAV?

mhkabir, 2016-10-05:
> armed the FCU the position didn't reset (x,y,z cant become 0). how can I fix it?

It is not supposed to. The local frame reference should never be reset.

meteor0530, 2016-10-05:
@tumbili  @julianoes 
I am sure im using INAV for estimate position. 
I use an older version of QGC (2.2.0) to download firmware. After arming in LOCAL_POSITION_NED x,y,z reset to 0.
[image]
[image]

AndreasAntener, 2016-10-05:
@meteor0530 this was the case in a (very) old firmware state but was found to be wrong and then fixed. If you are working with local position setpoints (offboard or on board) you don't want to handle a sudden reference reset on arming.

closed 2016-10-05 by AndreasAntener

meteor0530, 2016-10-05:
@AndreasAntener 
I am used to switch to OFFBOARD after arming, so I think reference reset before OFFBOARD is convenient.
If I move my UAV to another position, how can I define this posiiton as the new origin?

AndreasAntener, 2016-10-05:
It's easy to just keep your own reference from the point in time where you need it.