PX4/PX4-Autopilot #5608: Move channels to MAIN for Delta VTOL

sanderux, 2016-10-05:
I would propose to move the fixed wing channels to the MAIN output group for DeltaQuads.
This free's the AUX channels for passthrough and adds support for deltaquads on FC's without an AUX rail.

@LorenzMeier @AndreasAntener @tumbili @SimonWilks 
Any objections?

dagar, 2016-10-05:
Putting the fixed wing channels on MAIN sounds good to me, but which flight controllers have 7 MAIN outputs, but no AUX?

sanderux, 2016-10-05:
flytpod, luci, erlebrain
not sure about pixhawk2

kd0aij, 2016-10-05:
does MAIN imply channels implemented in the IO coprocessor?

dagar, 2016-10-05:
Yes, at least for the pixhawk. Having the px4io available for FW manual
passthrough and flight terminate is my interest.

kd0aij, 2016-10-05:
I don't think luci has an IO coprocessor, if my memory of the last devcall is correct

AndreasAntener, 2016-10-05:
FMUv4 also uses the mixers on MAIN outputs right? So MAIN doesn't imply IO. And other manufacturers use the terminology AUX (as in auxiliary) for "additional" outputs, like we discussed with Greg from Luci

AndreasAntener, 2016-10-05:
Moving all the outputs to MAIN by default would at least allow it to run on all boards with enough outputs, and gives Mark the override ;)

kd0aij, 2016-10-05:
Makes sense. We should be careful not to falsely imply the presence of a failsafe coprocessor (and the availability of MANUAL mode) if no coprocessor is present.

tubeme, 2016-10-10:
We've done this at 1.1 and no problems so far, very convenient to check and follow problems. We've done it because there was no way we could see the AUX buses in the log file. PX4 just did not log the AUX at that time. Runs almost on anything except PixRacer this way.

sanderux, 2016-10-10:
@tubeme could you share your change?

tubeme, 2016-10-10:
# VTOL quad X mixer for PX4FMU
#=============================

R: 4x 10000 10000 10000 0

Elevon mixers
-------------
Three scalers total (output, roll, pitch).

On the assumption that the two elevon servos are physically reversed, the pitch
input is inverted between the two servos.

M: 2
O:      10000  10000      0 -10000  10000
S: 1 0   7500   7500      0 -10000  10000
S: 1 1   6500   6500      0 -10000  10000

M: 2
O:       10000   10000    0 -10000  10000
S: 1 0    7500    7500    0 -10000  10000
S: 1 1   -6500   -6500    0 -10000  10000

Throttle mixer
--------------
M: 1
O:      10000  10000      0 -10000  10000
S: 1 3      0  20000 -10000 -10000  10000

Z:

that is in vtol_quad_x.main.mix

vtol_delta.aux.mix is empty.

May be there should be vtol_delta.main.mix

CH8 we use for servo power.

tubeme, 2016-10-14:
Forgot to mention important detail.

the quad is 400hz and the rest is 50hz.

config.txt

set MIXER vtol_quad_x
set PWM_OUT 12345678
set PWM_RATE 400
set PWM_DISARMED 900
set PWM_MIN 1100
set PWM_MAX 1900

extras.txt

pwm disarmed -c 56 -p 1500
pwm min -c 56 -p 1000
pwm max -c 56 -p 2000
pwm disarmed -c 78 -p 900
pwm min -c 78 -p 1000
pwm max -c 78 -p 2000
pwm failsafe -c 5 -p 1500
pwm failsafe -c 6 -p 1500
pwm failsafe -c 7 -p 1100
pwm terminatefail on

This is to guarantee the servos will stay in mid position when disarmed.

dagar, 2016-11-27:
So the main thing this is missing in the init scripts is having MAIN set multiple rates.
How would the IO manual override know to only use FW outputs?

tubeme, 2016-12-03:
@dagar With this scripts everything is perfect in Manual. Or probably I don't understand your point?

dagar, 2016-12-03:
My point is that if we're going to change one of the built in airframe configurations we need to make sure the PWM rates are set correctly. We can't have a built in configuration dependent on an additional sdcard script.

dagar, 2017-09-21:
Done

closed 2017-09-21 by dagar