PX4/PX4-Autopilot #5609: Descend mode when landed

mswingtra, 2016-10-05:
If there is no RC and gps connection (so possible workshop conditions), and aircraft is armed, then it will immediately go to auto descend mode. This essentially triggers landing on the current position. However, because aircraft is already on the ground, the ground detection in MPC fails since [it requires aircraft to move downwards firstly](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1722-L1727). 
Additionally, MPC will set the thrust setpoint to `MPC_THR_MIN` which is enough to spin up the props (dangerous on its own). With that, land detector will detect takeoff, and it will stay in this mode forever unless, for example, min. thrust is enough to cause tipping of the aircraft.

julianoes, 2016-10-11:
Interesting edge case, thanks for reporting.

mhkabir, 2017-04-03:
Is this still an issue? Our land detection system and controllers have undergone significant changes in the recent past.

julianoes, 2017-04-18:
@mhkabir let's close it. @mswingtra if this is still an isuse, please re-open.

closed 2017-04-18 by julianoes