PX4/PX4-Autopilot #5612: Cant run the 1st code on Snapdragon

anujkumargupta, 2016-10-06:
Hi
I am working with Snapdragon FC and trying to run my 1st code as mentioned in dev.px4.io.
I have build the code using make eagle_default and uploaded it in FC but when i am running the code found this error

root@linaro-developer:/home/linaro# ./px4 mainapp.config 
sh: 1: cannot create /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor: Directory nonexistent
commands file: mainapp.config
creating new parameters file
creating new dataman file

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
INFO  [sdlog2] [blackbox] /root/log/sess001
INFO  [sdlog2] [blackbox] recording: log001.px4log
ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
  MAV_BROADCAST: curr: 0ERROR [platforms__posix__px4_layer] krait update param 301 failed
 -> new: 1
  SDLOG_PRIO_BOOST: curr: 2ERROR [platforms__posix__px4_layer] krait update param 664 failed
 -> new: 3
INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 103090 bytes
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
pxh> INFO  [mavlink] using network interface wlan0, IP: 192.168.1.1
INFO  [mavlink] with broadcast IP: 192.168.1.255
anuj@anuj-vm:~/src/Firmware$ 

Earlier i was getting the same error so i had build all the cross toolchain installation and build the code using make but again same error.

Mini-DM is not showing anything

anuj@anuj-vm:~/src/Firmware$ ${HEXAGON_SDK_ROOT}/tools/debug/mini-dm/Linux_Debug/mini-dm
Running mini-dm version: 3.0
Device found with Product ID 0x9025. Continuing...
mini-dm is waiting for a DMSS connection...
DMSS is connected. Running mini-dm...

thanks

julianoes, 2016-10-06:
Have you updated to the BSP 3.1.1 and the latest adsp image?

Stifael, 2016-10-06:
same error, but also have timer difference too big

anujkumargupta, 2016-10-07:
Thanks @julianoes ,
Yes updated the linux image to BSP 3.1.1 and pushed the latest ADSP firmware , both downloaded from intrinsyc about 15 days ago.

julianoes, 2016-10-07:
@Stifael: does the error persist with thr latest BSP and adsp image?

julianoes, 2016-10-07:
@anujkumargupta and @Stifael: I can't reproduce the error, it works for me.

Can you list down all the commands you entered, maybe I can spot what you're doing different.

anujkumargupta, 2016-10-10:
HI @julianoes Today i just tried a different thing instead of pushing the q6-admin.sh i edited it in adb shell and now it works fine but still the Ground control station does detect it.
I tried to run mini-dm but didnot able to under stand

here is mini-dm output

anuj@anuj-vm:~/Qualcomm/Hexagon_SDK/3.0/tools/debug/mini-dm/Linux_Debug$ ./mini-dm 
Running mini-dm version: 3.0
Device found with Product ID 0x9025. Continuing...
mini-dm is waiting for a DMSS connection...
DMSS is connected. Running mini-dm...
[08500/00]  02:33.700  HAP:16415:In dspal_entry  0222  main.cpp
[08500/00]  02:33.700  HAP:16415:92813238 WorkMgr already initialized   0063  WorkMgr.cpp
[08500/02]  02:33.710  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.710  HAP:28713:92823368 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.710  HAP:28713:92823416 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.710  HAP:28713:92823450 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.710  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.713  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.713  HAP:28713:92827381 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.715  HAP:28713:92827427 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.715  HAP:28713:92827461 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.715  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.718  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.718  HAP:28713:92831499 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.718  HAP:28713:92831541 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.718  HAP:28713:92831577 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.718  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.722  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.722  HAP:28713:92835739 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.722  HAP:28713:92835784 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.722  HAP:28713:92835821 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.722  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.730  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.730  HAP:28713:92843155 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.730  HAP:28713:92843199 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.730  HAP:28713:92843237 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.730  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.733  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.733  HAP:28713:92847009 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.733  HAP:28713:92847052 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.733  HAP:28713:92847086 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.733  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  02:33.738  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.738  HAP:28713:92851850 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.738  HAP:28713:92851893 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.738  HAP:28713:92851931 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.738  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/00]  02:33.741  HAP:16415:attempting to open the ADSP command file: /dev/fs/px4.config  0186  main.cpp
[08500/02]  02:33.742  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:33.742  HAP:28713:92855681 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  02:33.742  HAP:28713:92855721 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  02:33.742  HAP:28713:92855758 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  02:33.742  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/00]  02:33.745  HAP:16415:reading commands from /dev/fs/px4.config   0193  main.cpp
[08500/02]  02:33.746  HAP:16415:already loaded  0067  uORBMain.cpp
[08500/00]  02:33.746  HAP:16415:already running  0081  qshell_start_qurt.cpp
[... cut ...]
[08500/00]  02:36.017  loading the blsp config file failed, loading the default config  0261  main.c
[08500/02]  02:36.017  Loaded default UART-BAM mapping  0036  blsp_config.c
[08500/02]  02:36.017  UART tty-1: BAM-9  0044  blsp_config.c
[08500/02]  02:36.017  UART tty-2: BAM-6  0044  blsp_config.c
[08500/02]  02:36.017  UART tty-3: BAM-8  0044  blsp_config.c
[08500/02]  02:36.017  UART tty-4: BAM-2  0044  blsp_config.c
[08500/02]  02:36.017  UART tty-5: BAM N/A  0049  blsp_config.c
[08500/02]  02:36.017  UART tty-6: BAM N/A  0049  blsp_config.c

Thanks

julianoes, 2016-10-10:
Do you always get this `FIFO overflow` output, even after a reboot and a power-cycle (disconnect all power)?
Please post the output of the shell and mini-dm. Also, is the networking setup correctly? Can you ping the device?

anujkumargupta, 2016-10-10:
Hi 
This adb shell output

root@linaro-developer:/home/linaro# ./px4 mainapp.config 
sh: 1: cannot create /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor: Directory nonexistent
commands file: mainapp.config

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [sdlog2] [blackbox] /root/log/sess009
INFO  [sdlog2] [blackbox] recording: log001.px4log
INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 103090 bytes
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
pxh> INFO  [mavlink] using network interface wlan0, IP: 192.168.1.1
INFO  [mavlink] with broadcast IP: 192.168.1.255

This is mini-dm

anuj@anuj-vm:~/Qualcomm/Hexagon_SDK/3.0/tools/debug/mini-dm/Linux_Debug$ ./mini-dm
Running mini-dm version: 3.0
Device found with Product ID 0x9025. Continuing...
mini-dm is waiting for a DMSS connection...
DMSS is connected. Running mini-dm...
[08500/00]  22:55.782  HAP:61:In dspal_entry  0222  main.cpp
[08500/00]  22:55.802  HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config  0186  main.cpp
[08500/00]  22:55.802  HAP:61:reading commands from /dev/fs/px4.config   0193  main.cpp
[08500/02]  22:55.920  HAP:61:Sleeping for 1, 1000000  0118  apps.h
[08500/00]  22:56.921  configuring UART for 4-wire mode, DAL id: 0x2001001  0852  DalUart.c
[08500/02]  22:56.926  HAP:61:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  22:56.926  HAP:61:1143820 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  22:56.926  HAP:61:1143862 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  22:56.926  HAP:61:1143895 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  22:56.926  HAP:61:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/03]  22:56.926  HAP:61:DfHmc9250Wrapper start failed  0347  df_hmc5883_wrapper.cpp
[08500/00]  22:57.030  HAP:61:1246345 Reset MPU9250   0089  MPU9250.cpp
[08500/00]  22:57.093  HAP:60:1310597 IMU temperature initialized to: 50.468143   0516  MPU9250.cpp
[08500/00]  22:57.095  HAP:61:1312495 additional sensor configuration succeeded   0187  BMP280.cpp
[08500/03]  22:57.096  HAP:61:start  0204  df_trone_wrapper.cpp
[08500/02]  22:57.098  HAP:61:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  22:57.098  HAP:61:1315465 Error: i2c write failed. Reported -1 bytes written   0216  I2CDevObj.cpp
[08500/00]  22:57.098  HAP:61:1315498 Nonzero result when writing to TRONE_WHO_AM_I_REG 0x1   0110  TROne.cpp
[08500/00]  22:57.098  HAP:61:1315535 probing not successful   0071  TROne.cpp
[08500/03]  22:57.098  HAP:61:TROne start fail: -1  0134  df_trone_wrapper.cpp
[08500/03]  22:57.098  HAP:61:DfTROneWrapper start failed  0215  df_trone_wrapper.cpp
[08500/02]  22:57.105  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  22:57.105  HAP:28713:1322879 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  22:57.105  HAP:28713:1322921 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  22:57.106  HAP:28713:1322954 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  22:57.106  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  22:57.107  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  22:57.107  HAP:28713:1324667 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  22:57.107  HAP:28713:1324706 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  22:57.107  HAP:28713:1324739 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  22:57.107  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  22:57.110  HAP:12328:LED: getHandle fail   0269  commander_helper.cpp
[08500/02]  22:57.110  HAP:12328:LED init failed  1287  commander.cpp
[08500/02]  22:57.110  HAP:12328:Buzzer: px4_open fail   0132  commander_helper.cpp
[08500/02]  22:57.110  HAP:12328:Buzzer init failed  1291  commander.cpp
[08500/02]  22:57.110  HAP:12328:Preflight checks always pass on Snapdragon.  0392  PreflightCheck.cpp
[08500/00]  22:57.175  configuring UART for 4-wire mode, DAL id: 0x2001005  0852  DalUart.c
[08500/00]  22:57.175  HAP:35:Trying to initialize mixer from config file /dev/fs/quad_x.main.mix  0145  pwm_out_rc_in.cpp
[08500/00]  22:57.177  HAP:35:Successfully initialized mixer from config file  0157  pwm_out_rc_in.cpp
[08500/02]  22:57.177  HAP:28713:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  22:57.177  HAP:28713:1394489 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  22:57.177  HAP:28713:1394533 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  22:57.177  HAP:28713:1394571 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  22:57.177  HAP:28713:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/00]  22:58.748  HAP:38:EKF using pressure height - commencing alignment  0407  ekf.cpp
[... cut ...]
i tried to connect it using wifi module but the signal is very week (even when i kept it at a distance of max 20cm) and ping output shows destination host not reachable 

root@linaro-developer:/# ping 192.168.1.12
PING 192.168.1.12 (192.168.1.12) 56(84) bytes of data.
From 192.168.1.1 icmp_seq=1 Destination Host Unreachable
From 192.168.1.1 icmp_seq=2 Destination Host Unreachable
From 192.168.1.1 icmp_seq=3 Destination Host Unreachable
^C
--- 192.168.1.12 ping statistics ---
6 packets transmitted, 0 received, +3 errors, 100% packet loss, time 5006ms
pipe 3

How to increase its signal strength

julianoes, 2016-10-10:
Do you have an antenna connected to the Snapdragon? This would increase the signal strength. However, for testing I use it without antenna and it is usually good enough for a few meters.

julianoes, 2016-10-10:
@anujkumargupta it looks like the first error is gone though:
```
[platforms__posix__px4_layer] krait update param 664 failed
```

Do you know what you changed to resolve it?

anujkumargupta, 2016-10-10:
Hi earlier in pinging it shows error because i also connected my lan cable but after removing the lan i can ping the device.

root@linaro-developer:/# ping 192.168.1.12
PING 192.168.1.12 (192.168.1.12) 56(84) bytes of data.
64 bytes from 192.168.1.12: icmp_seq=1 ttl=64 time=4.90 ms
64 bytes from 192.168.1.12: icmp_seq=2 ttl=64 time=11.3 ms
64 bytes from 192.168.1.12: icmp_seq=3 ttl=64 time=23.0 ms
64 bytes from 192.168.1.12: icmp_seq=4 ttl=64 time=7.85 ms
64 bytes from 192.168.1.12: icmp_seq=5 ttl=64 time=5.88 ms

but the adb shell and mini-dm output is same
Thanks

julianoes, 2016-10-10:
Ok, good so it was a routing issue.

The adb shell and mini-dm output look ok. It is just complaining that no external mag is connected but it will fall back to the internal mag.

I would expect that QGC shows something now, no?

anujkumargupta, 2016-10-10:
Hi @julianoes 
Yes this [platforms__posix__px4_layer] krait update param 664 failed error is solved.
Earlier i was trying to pull the script and then edit and then pushing it.
adb pull /usr/local/qr-linux/q6-admin.sh
gedit q6-admin.sh
adb push q6-admin.sh /usr/local/qr-linux/q6-admin.sh
But later i edited it directly through adb shell
vim /usr/local/qr-linux/q6-admin.sh
and also done all step without removing the board even once.

julianoes, 2016-10-10:
Ok interesting, I'll try to reproduce this. In theory both "editing" variants should lead to the same.

julianoes, 2016-10-17:
@anujkumargupta you were right! If you do `adb pull` and `adb push` the executable bit gets lost.
Fixed with https://github.com/PX4/Devguide/pull/55.

closed 2016-10-17 by LorenzMeier