PX4/PX4-Autopilot #5621: Interfacing the Simulation to ROS gazebo error

kiti1013, 2016-10-08:
I follow the developer guide to run the simulation，when I run the command 

`roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/iris.world
`
 the terminal prompt 
[gazebo-1] process has died [pid 4680, exit code 134, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver -e ode /home/astera/Tools/sitl_gazebo/worlds/iris.world __name:=gazebo __log:=/home/astera/.ros/log/6e8dd0d8-8d26-11e6-845f-8c89a509c310/gazebo-1.log].
[gazebo-1] process has died [pid 4680, exit code 134, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver -e ode /home/astera/Tools/sitl_gazebo/worlds/iris.world __name:=gazebo __log:=/home/astera/.ros/log/6e8dd0d8-8d26-11e6-845f-8c89a509c310/gazebo-1.log].
[gazebo-1] process has died [pid 4680, exit code 134, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver -e ode /home/astera/Tools/sitl_gazebo/worlds/iris.world __name:=gazebo __log:=/home/astera/.ros/log/6e8dd0d8-8d26-11e6-845f-8c89a509c310/gazebo-1.log].

my os  ubuntu 14.04 
    ros  indigo
   gazbo 6.6 （sudo apt-get install ros-indigo-gazebo6-ros）

julianoes, 2016-10-11:
Hi @kiti1013,
any chance you get more output on why they all die? Are you able to use the normal gazebo simulation without ROS using `make posix gazebo`?

kiti1013, 2016-10-13:
i can  use gazebo simlation normalily without ros

julianoes, 2016-10-13:
@kiti1013: try running one of the commands that fail on it's own, presumably do:
```
roscore
```

and then
```
opt/ros/indigo/lib/gazebo_ros/gzserver -e ode /home/astera/Tools/sitl_gazebo/worlds/iris.world __name:=gazebo __log:=/home/astera/.ros/log/6e8dd0d8-8d26-11e6-845f-8c89a509c310/gazebo-1.log
```

or check the launch file for what it starts. This should give you more output on what's happening.

closed 2017-03-01 by kiti1013

Hanslen, 2018-09-30:
I have the same problem, is there any solution?

zhaoxiaowei1013, 2018-09-30:
@Hanslen do you what to run a muti copter to simulation？

julianoes, 2018-09-30:
@Hanslen this is an old thread. Please re-open a new issue and add detailed a description about what your issue is.