PX4/PX4-Autopilot #5624: No response to aileron RC input in ALTCTL mode

kd0aij, 2016-10-10:
http://logs.uaventure.com/view/AqovJ2XamZWcXUvv7d32UG#AS_Roll_PLOT
Firmware v1.4.4

kd0aij, 2016-10-11:
It looks like this commit (from 20 July) is at least part of the problem: https://github.com/PX4/Firmware/commit/aea7bd5b471cd64ddef843fe0c912f6b0e2eb679

Problem is that attitude setpoints are published in fw_att_control only 
```
if (!_vcontrol_mode.flag_control_climb_rate_enabled)
```
and this condition is not met for ALTCTL.

Perhaps the intended control mode flag is flag_control_position_enabled?

dagar, 2016-10-11:
For fixed wing AUTO, POSCTL, ALTCTL attitude setpoints are published by fw_pos_ctrl_l1, with TECS responsible for throttle and pitch.

Could you explain the issue a little more? Is this a logging issue, or in altitude control there's no roll at all? What happens in the other modes? AUTO, POSCTL, STAB

kd0aij, 2016-10-11:
As the title of the issue suggests: "No response to aileron RC input in ALTCTL mode" by which I mean you can't turn. I don't think I tried rudder, but that wouldn't be very useful on flying wings.

The log shows that STAB and POSCTL are OK, though I can't find any documentation on the functions of throttle and elevator in POSCTL. I assume that AUTO modes are supposed to ignore RC controls.

kd0aij, 2016-10-11:
@dagar A grep for "flag_control_climb_rate_enabled" didn't get any hits in fw_pos_ctrl_l1.
As far as I can tell, climb rate control isn't implemented anywhere.

kd0aij, 2016-10-11:
@dagar Do you use Eclipse? For some reason, the C++ index seems to be screwed up with this release, because e.g. when I search for "flag_control_climb_rate_enabled" I get bogus matches in mc_pos_control_main.cpp (the line numbers are all wrong, off by as much as 10 lines)
I've tried a complete out-of-tree rebuild, configure, generate, run indexer, but that didn't help.

dagar, 2016-10-12:
I committed my eclipse changes here - https://github.com/PX4/Firmware/pull/5646
Still far from perfect, but it properly finds everything for the main pieces (commander, controllers, etc).

kd0aij, 2016-10-12:
@dagar I've been using the cmake project generator for a long time, but recently it only works for posix builds:
```
rm -rf build_firstPX4
mkdir build_firstPX4
cd build_firstPX4
cmake -G Eclipse\ CDT4\ -\ Ninja ../Firmware/
```

kd0aij, 2016-10-12:
@LorenzMeier @dagar This seems to be badly broken: https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L959

In STABILIZE mode, this module publishes roll/pitch/yaw/thrust setpoints because flag_control_attitude_enabled == true and flag_control_climb_rate_enabled == false. That's good. 

But in ALTCTL mode, flag_control_climb_rate_enabled is true, and no attitude setpoint is published here. That's bad, because then nobody is publishing the roll/yaw setpoints and you can't turn.
Changing the test to flag_control_position_enabled==false would still work for stabilize mode, but not for altctl, because TECS is supposed to be handling throttle and pitch. As far as I know, you can't have one module publishing some of the fields in the attitude_setpoint topic, with another module publishing the rest.

dagar, 2016-10-12:
In ALTCTL the attitude setpoint is published here - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L2013

flag_control_climb_rate_enabled is on with flag_control_altitude_enabled.

I'll take a look at your log.

kd0aij, 2016-10-12:
@dagar Thanks, don't know why my code searches didn't find that, but I was probably looking for control_climb_rate_enabled, since that is the test in att_control. Hard to explain why my log shows no response to the RC roll channel in ALTCTL too.

dagar, 2016-10-12:
Okay, so the problem is lack of GPS. The fixedwing position controller only runs with an updated global position. https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L2280

Modifying ALTCTL to work without GPS would be nice.

kd0aij, 2016-10-12:
That must have been the flight with the GPS plugged into the wrong port :)
One might expect the autopilot to update RC commanded setpoints at 50Hz, regardless.

dagar, 2016-10-13:
This is actually pretty terrible. We either need to look at fixing fw_pos_ctrl_l1 or the state machine.

kd0aij, 2016-10-13:
How about running the main loop at 50Hz and checking for changes in either position or RC inputs?

dagar, 2016-10-13:
Altitude control mode is currently using _global_pos.alt, so we'd have to fallback to something else. The do_takeoff_help() is also a problem without GPS.

https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L2058

tumbili, 2016-10-13:
@dagar I believe I opened an issue for this not too long ago, will check.

dagar, 2016-12-04:
Closing as duplicate of https://github.com/PX4/Firmware/issues/5123

closed 2016-12-04 by dagar