PX4/PX4-Autopilot #5625: 1.5.0 RC1 is reported as 1.4.4 rc1

tubeme, 2016-10-10:
In the logs and in QGroundControl.

Flashed 1.5.0 RC1 many times to various HW. The same story.

DonLakeFlyer, 2016-10-10:
I thought @dagar was saying this was a url problem to the stable builds. But I also though that he said he repushed them and fixed it.

dagar, 2016-10-11:
I'll try to figure out what's going on.

dagar, 2016-10-11:
Sigh, let me guess px4fmu-v4?

closed 2016-10-11 by dagar

dagar, 2016-10-11:
This was a completely different problem, hidden beneath the previous problem. PX4 master fixed, beta will be fixed shortly.

tubeme, 2016-10-11:
It is px4fmu-v2.

dagar, 2016-10-11:
So px4fmu-v4 was definitely wrong because it wasn't uploading. Let me see if there's yet another issue with the versioning itself.

davids5, 2016-10-11:
> This was a completely different problem, hidden beneath the previous problem. PX4 master fixed, beta will be fixed shortly.

@dagar What is the root cause or commit that fixed this?

dagar, 2016-10-16:
@davids5 - https://github.com/PX4/Firmware/commit/e9e15ba3d1d3ae954dc5e27e1c01b50e63f06ca8#diff-b67911656ef5d18c4ae36cb6741b7965

Before that someone hadn't broken travis-ci uploads.

PR #5635 (2016-10-11): Makefile error preventing px4fmu-v4_default build
 -fixes #5625
commit e9e15ba3 (2016-10-11): Makefile error preventing px4fmu-v4_default build

 -fixes #5625
PR #5636 (2016-10-11): Makefile error preventing px4fmu-v4_default build
 -fixes #5625
 -cherry-pick master -> beta
commit d0afcd1b (2016-10-11): Makefile error preventing px4fmu-v4_default build

 -fixes #5625