PX4/PX4-Autopilot #5626: VTOL: 1.5.0 RC1 Back Transition problems.

tubeme, 2016-10-10:
As discussed before, our opinion is that for the VTOL back transition phases we should apply the rules for the classic airplane approach and landing. 

So now in 1.5 we have the flare or the quad - air-baking - maneuver after we switch back transition and it is very, very good thing in order to brake properly the speed. But in our case it almost brakes the plane. It gives such a sharp almost 40deg pitch up punch that the plane is experiencing very strong vibrations. We joke that if there are people inside the plane they will throw up on the spot. :) Our test mprototypes are very stiff and strong, because we make them to handle excessive forces. But this will brake on the spot a regular EPO model.

The problem is not the degree of the pitch up, but rather the speed and the sharpness. It is good idea if we can control 3 things for this quad-flare. - The degree, the speed and the timing. 

It could be achieved introducing 3 VT_QUAD_FLARE parameters. Pitch Up Angle, Time to reach the angle, time to reach back to horizontal hover flight.

Another issue for reporting is the back transition timing in Mission. During mission back transition, the plane always overshoots the landing point with about 10-20 meters depending on the wind conditions, and then it comes back in quad mode and lands. For the plane going back it is not a good thing. So it is always better to make the transition before the landing spot, taking in account the airpseed and wind and then move forward in quad mode until prcisely reaching the spot and land. We discussed with @sanderux he told me it will be implemented. I reported this because it is still present in 1.5.0 RC1.

tubeme, 2016-12-03:
We have really huge problem with this air-braking. We've had a couple of flips during this maneuver especially when we lowered the weight of our frame with 300 grams. In a gusty weather it is almost not possible to make a good back transition. this is the 1.5 RC1. Today we will upgrade to RC3, hope everything is correct there.

dagar, 2016-12-04:
@tubeme I think what we really need here is the VTOL architecture change described here - https://github.com/PX4/Firmware/issues/4845

Then we can do more with the position controllers, potentially including things like in air flare to brake (while controlling airspeed), and properly use then have the multicopter do the rest.

The back transition timing is primarily that we don't have a good understanding of how long/far a back transition will take, including dependence on airspeed, groundspeed, and vehicle specifics.

dagar, 2017-04-30:
@tubeme have you tried this recently with master than maintains FW pitch setpoint offset during back transition? Next step will be using the position controllers for active braking. https://github.com/PX4/Firmware/issues/6856

closed 2017-04-30 by dagar