PX4/PX4-Autopilot #5627: Barometric Altimeter Calibration.

tubeme, 2016-10-10:
Part of our flights are at see level. We have a lot of logs like this.

[image]

If PX4 is on the ground and has a GPS lock, the GPS altitude reading should be pretty stable and correct, given the machine is not moving. This could be used to some how calibrate the barometric sensor, prior flight. Or at least give us the option to calibrate it manually.

AndreasAntener, 2016-10-10:
Try SENS_BARO_QNH (https://en.wikipedia.org/wiki/QNH)
And you should be looking at GPOS.Alt, not the raw baro alt (which is btw more stable than your GPS altitude that is clearly drifting on the ground).

closed 2016-10-10 by AndreasAntener

tubeme, 2016-10-11:
Yes it is the QNH we have to adjust. But I thought there could be another way for calibration.

I prefer to see raw data from the sensors during our tests. That is why I look at Barometric sensor data rather than GPOS.Alt 

And you are correct the GPS alt is drifting and not stable for calibration purposes. Probably the RTK GPS systems will allow this to happen because they are very precise in 3d space.