PX4/PX4-Autopilot #5628: Read UART inside a hrt_call

FAC94, 2016-10-10:
Hello,
I'm trying to make an IMU drive to a sensor that communicates through UART with the autopilot and works as follow:
1) When a read is wanted, it is set a pin to request a measure of the sensor.
2) Then it answers through UART a fixed size message.

My problem is that I'm not able to make a read of the UART inside an hrt_call function. How can I overcome this?

Thanks

julianoes, 2016-10-10:
Hi, I think you should only use the hrt_call for something that absolutely needs to run regularly such as a gyro or accel driver. Reading from a UART is usually done in a NuttX task in a while loop with some usleep, or using the low or high priority work queue.

FAC94, 2016-10-10:
Hi @julianoes 
Thanks for your answer.

The problem is exactly that, the sensor that I'm talking provides gyro and accel data, that's why I was using the hrt_call (similar to the SPI sensors drivers).

However I can only read its data from an UART. 

I tried to make a task as you said that would spwan when I give order to the sensor start (like the other sensors in rc.sensors file). However, that way, the system is not capable of performing the complete start and blocks the NSH, which leads to a bad initialization of the system.

davids5, 2016-10-10:
You will need to mimic a driver - with a main entry that spawns a task.

FAC94, 2016-10-10:
Ok, so now I'm able to do a proper start, but I still have the problem in reading the serial port inside the task. 

The serial port is opened with the identifier _serial_fd and it is working fine (I can read properly out of the hrt_cal and the task_spawn).

Basically the idea was to create a task to do readings, while the measure cycle is responsible for only fill the structures and release a semaphore (for trying to get a measure period as constant as possible).

Here is a piece of the code that relates to that part:
```

void SENSOR::start()
{
    _accel_reports->flush();
    _gyro_reports->flush();

    hrt_call_every(&_call,
                   1000,
                   _call_interval,
                   (hrt_callout)&SENSOR::measure_trampoline,this);

    sem_init(&_ru_semaphore, 0, 0);

    _task = task_spawn_cmd("SENSOR", SCHED_DEFAULT,
                SCHED_PRIORITY_SLOW_DRIVER, 1500, (main_t)&SENSOR::read_uart_tramp, nullptr);
}

void SENSOR::read_uart_tramp(void *arg)
{
    SENSOR *dev = (SENSOR *)arg;
    dev->read_uart();
}

void SENSOR::measure_trampoline(void *arg)
{
    SENSOR *dev = (SENSOR *)arg;
    dev->measure();
}

void SENSOR::read_uart()
{
    while(true)
    {
        sem_wait(&_ru_semaphore);
        stm32_configgpio(GPIO_SENSOR_EXT_TRIG);
        stm32_gpiowrite(GPIO_SENSOR_EXT_TRIG, 0);
        usleep(100);
        int fd = _serial_fd;

        ::read(fd,buf_read,sizeof(buf_read));

        stm32_gpiowrite(GPIO_SENSOR_EXT_TRIG, 1);
    }
}

void SENSOR::measure()
{
    //CODE to fill the structures
    
    sem_post(&_ru_semaphore);
}
```

davids5, 2016-10-10:
1) Do not call read in the HRT - it is running off the timer ISR and can not use any resources.
2) Open the FD in the task's context.

more like this:
```
task_main:
   _serial_fd= open(...)
   stm32_configgpio(GPIO_SENSOR_EXT_TRIG);
     
   while(task_run) {

         sem_wait(&_ru_semaphore);
        stm32_gpiowrite(GPIO_SENSOR_EXT_TRIG, 0);
        usleep(100);
        ::read(_serial_fd,buf_read,sizeof(buf_read)); // Need a proper read with length
        stm32_gpiowrite(GPIO_SENSOR_EXT_TRIG, 1);
   }
close(_serial_fd);
```

FAC94, 2016-10-10:
@davids5 Thanks for your help! It seems to be working now :D
Although I was opening the serial port earlier in the code, it seems that I needed to open it in the "task_main" function as you suggested.

julianoes, 2016-10-11:
@FAC94 please close the issue if it is resolved, thank you.

closed 2016-10-11 by FAC94