PX4/PX4-Autopilot #5629: MC attitude septoints are zero when motors are saturated. v 1.4.4

ilia-sheremet, 2016-10-10:
I experienced several times while flying hex copter in wind conditions around 6 m/s:
12_08_23.px4log
[image]

In PosControl the copter saturates from around 84s till 88s. You can see how desired pitch (and roll as well, you could see from logs) jumps from stabilizing copter to zero. Sometimes (86.5s - 87s) you can see that desired pitch goes back to stabilize when the motors are still saturated, but I believe it might be somewhere on the boundary of saturation and it unblocks the attitude control.

But it seems that copter will start to stabilize attitude even when the motors are saturated if you give an rc input, like you can observe on the picture below. At around 270.5 it goes from off-board to poscontrol mode while motors are full up and only 272, when the rc input for position change are given the desired attitude starts to be non-zero.
10_03_49.px4log
[image]

If it's intentionally made that multirotors don't keep the attitude when the motors are full powered in order to keep altitude, I don't think it's the best idea. When the copter start to suddenly fly away without you commanding it and completely neglecting wind in my opinion more dangerous if it would slowly go down while still trying to fight the wind.    

[Link](https://drive.google.com/drive/folders/0B4HSR_JQFK8KRVhJNVpHQ2NJZFk?usp=sharing) to px4logs.

devbharat, 2016-10-11:
I think mpc is implemented to still control the attitude, just not the position, setting the pitch&roll setpoints to zero to hold altitude. 

It's implemented [here](https://github.com/PX4/Firmware/blob/d80a233bcb88c3839d7ac2d7089a8156820834df/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1815-#L1822).

ilia-sheremet, 2016-10-28:
@devbharat Thank you for the reply, that's what I thought as well.

But don't you think it's a wrong approach to handle the condition? @LorenzMeier @tumbili @AndreasAntener 

I think in most of the cases this approach has higher probability to induce a crash. In my case it's slightly underpowered copter with hover at 60% thrust, flying in Position Mode with moderate wind. It's successfully fighting the wind and than abruptly without any messages/indicators/switching_modes levels up and rapidly flows away with the wind, responding badly to control signal. I was lucky to not to have any building or trees next to me, but I would fly close to obstacle the chance of me reacting fast enough would be low.

And please consider inexperienced users who could barely fly with position hold. For them this situation and copter being oriented towards them might cause panic. They might not even think to switch into ALT hold and get the copter into position and with Position mode it would be just gradually floating away from them.

IMHO keeping the XY position should have higher priority above altitude in this case. A copter might fight the wind and gradually descending (due to the lack of power). Anyway this descending will be slow and controllable in most of the cases, but it's much more intuitive and visible for the pilot that the copter doesn't have enough power, than just leveling up and flying away. (without visible reason for the user)

mhkabir, 2017-04-03:
Still and issue?

ilia-sheremet, 2017-04-03:
@mhkabir Haven't checked that with the latest firmware since it's not that easy to reproduce it. But I guess the behavior is the same if nobody worked on that

LorenzMeier, 2017-04-12:
Fixed

closed 2017-04-12 by LorenzMeier