PX4/PX4-Autopilot #5631: Fix preflightCheck calls and remove default value for last argument

AndreasAntener, 2016-10-10:
- remove default argument here: https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.h#L70 (this was already the reason for the last preflight check bugs)
- fix calls in commander

sanderux, 2016-10-11:
@AndreasAntener i don't believe i changed this. do you need the default removed for reportFailures?

sanderux, 2016-10-11:
The default was added in this commit: https://github.com/PX4/Firmware/commit/5fcfdb759c2a83c3e13235b0f71d8d6299d46c33#diff-390f83af28dcf517efab5f9cb487194cR70

AndreasAntener, 2016-10-11:
> @AndreasAntener i don't believe i changed this. do you need the default removed for reportFailures?

> c4eabbd#diff-a2bcb238e25d8a97ab2aee48f54dbc48R386

Yes, the default should be removed (and yes that wasn't you ;)), but your change missed to update all preflightCheck calls which are now wrong because of the new argument and the fact that the last argument has a default value masks that problem.

PR #5851 (2016-12-02): Airspeed: preflight check for bad offset, fixed calls to preflight ch…
…ecks (vtol & airspeed)

This is best used with a pitot tube cover. Without a cover and lots of wind on the ground the check will report a calibration issue. Although it's annoying to do more airspeed calibrations, this effectively got rid of all offset issues for us.

Fixes #5059 (I think)
Fixes #5631

@dagar @julianoes please have a look
closed 2016-12-02 by LorenzMeier