PX4/PX4-Autopilot #5632: loss of manual_control (loss of rc)

gregd72002, 2016-10-10:
Scenario: flying using manual_control mavlink messages

When the ground station stops to send manual_control the rc link should go into failsafe - at the moment it is being kept active all the time resulting in fly-away.

From my investigations:
- mavlink_receiver (MavlinkReceiver::handle_message_manual_control) - only publishes rc input values when manual_control message arrives (it never sets rc.rc_lost to true)
- Sensors::rc_poll() expects rc.rc_lost to be set in order to trigger failsafe

I can think of 2 solutions:
- MavlinkReceiver to implement some kind of timer on manual_control, after which rc is published with rc_lost=true (this would only be active in certain COM_RC_IN_MODE)
- Sensors::rc_poll() to keep track of last updated rc_input and to fallback into failsafe after timeout

Any views on this?

julianoes, 2016-10-10:
Thanks for the write-up, I would vote for:
> Sensors::rc_poll() to keep track of last updated rc_input and to fallback into failsafe after timeout

julianoes, 2016-10-10:
Related to #4876.

gregd72002, 2016-10-11:
I gave it a try and the only change needed would be to modify sensors.cpp::rc_poll()
- introduced timeout for rc_updated after which signal_lost is set
- common code within the `if (rc_updated) {` has been moved after the condition.

https://github.com/gregd72002/Firmware/commit/2dca552113c7ab8f49357d0aa5ef5b7926c49a66

Any comments?

gregd72002, 2016-10-11:
I've bench tested it. Pull request created: https://github.com/PX4/Firmware/pull/5641

bkueng, 2016-10-12:
I think this needs further investigations. From what I see `rc_channels` is just an output topic, and the `signal_lost` flag in sensors is not really relevant. sensors.cpp publishes `manual_control_setpoint`, which is handled in commander. And then commander checks for a timeout at https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2450 and handles failsafe (you should see a 'MANUAL CONTROL LOST' message).
The question then is: why does commander not go into failsafe mode?

gregd72002, 2016-10-12:
@bkueng in this scenario `manual_control_setpoint` is never published as such `sp_man` get zero'ed and the commander does not have a valid `sp_man.timestamp`. 
The check you are referring to (the else condition) should fire which would be ok. However, there is another check on `status.rc_signal_lost` which will be set to true as per default (L:1323), hence MANUAL CONTROL LOSS will not happen.

Could you confirm this is what you see too and we can take it from there.

bkueng, 2016-10-12:
Yes that all makes sense if `manual_control_setpoint` is never published at all. 
As you wrote:
> When the ground station stops to send manual_control the rc link should go into failsafe - at the moment it is being kept active all the time resulting in fly-away.

Which would mean that at some point (in the begining) there should be a valid `manual_control_setpoint` published. Is that not the case in your scenario? Am I missing something?

gregd72002, 2016-10-12:
you are right on it. I meant exactly this - `manual_control_setpoint` stops to be published.
I.e. the copter is armed (`manual_control_setpoint` is published), then consecutively `manual_control_setpoint` stops.

bkueng, 2016-10-12:
Ok, but then the MANUAL CONTROL LOST should be triggered in commander, and `status.rc_signal_lost = false;` must be executed before. Did you change the `COM_RC_LOSS_T` parameter?

gregd72002, 2016-10-12:
Right, I'm with you. Strange... I will need to do some more testing around it.

I did change `COM_RC_LOSS_T` to something like 3s. Does this matter?

bkueng, 2016-10-12:
No that's ok, but if you had it set to something large, like 100, it would explain things.

gregd72002, 2016-10-12:
I run some additional test around different versions and I cannot reproduce this on the recent development version. Though it is a case on stable release. Hence, not applicable anymore.

closed 2016-10-12 by gregd72002