PX4/PX4-Autopilot #5637: VTOL / FW : 1.5.0 RC1 too high angular roll speed in mission.

tubeme, 2016-10-11:
So far we've had more than a 100 missions in VTOL. About 70 on 1.3.4 and just recently about 30 missions in 1.5.0RC1.

Paralel to this we are flying regular FW and have pretty decent amount of missions.

In all this missions and flying we could not find a way to control the angular speed for the roll.

During autonomous flight (RTL, Mission) the plane is banging so hard in the turns that it almost brakes the wing. We make our prototypes very strong and stiff with carbon and composites, but if it is something softwe EPO, EPS it will not hold on this turns. I don't know how @sanderux X8 VTOL holds in this turns?

And here I want to clarify, the problem is not the bank angle but the speed and thus the force with which PX4 commands the plane to roll. 

In Position and Altitude we control this speed ourselfs but in Fully autonomous flight we have no control over it. Something like this: MC_YAWRAUTO_MAX for the MC.

If I cut the overall roll rate FW_R_RMAX then I loose the ability of the plane to react harder at strong weather situations and still perform a gentle turns in mission. 

We tried larger acceptance radius missions but it had no effect on the roll speed.

Any ideas how can we make it turn gentle in mission and RTL? but still maintain its ability to react harder on external forces?

tubeme, 2016-10-11:
Another example i can give is from last year tests.

The plane was upper wing, Cessna style IC engine airplane. 2.5m wingspan. ~20-23 m/s cruising speed.

Upper wing airplane have a tendency to dive roll when you make a sharp yaw turn with the rudder. 

So we set up coordinated turn after 20 m/s reached and open-loop method. Flying in Position and Altitude was pretty good given the coordinated turn.

Bu then we hit Loiter and something very strange happened. The plane roll dived recovered and started  Loiter. At first we thought it was a glitch. But then it happened again and again. It happened in RTL. First it roll dived in rtl in the initial turn, it recovered and came back, roll dived at the start of the Loiter, then recovered and started Loiter...

So we found the reason. It banged so much the rudder in the initial coordinated turn (either Loiter or RTL) that it almost roll dives the plane, then the tecs recover the plane and once bellow the 20 m/s it performed perfect. So the problem was the initial Rudder command. And again we could not control the roll rate speed in auto mode without sacrificing the plane's ability to fight strong weather. 

So we just turned off the coordinated turn. It is petty because coordinated turn saves energy in long missions.

sanderux, 2016-10-11:
@tubeme as with any reference to a flight not to your satisfactory; please provide a log

as for the roll behavior. you can lower the FW_RR_P
as for the sudden roll, please upload a log

sanderux, 2016-10-11:
As for my flight behavior, see for your self: 
https://drive.google.com/open?id=0B3otHVwcj_LdUmFEOHdLcFJ1VVE

tubeme, 2016-10-11:
Thanks for the Video. There I see a circular mission without sharp turns which diminishes the effects I'm talking about.

Here is the example.

[image]

[image]

And the Log:
http://logs.uaventure.com/view/szHmDnbEvtBKnGB8bfx4Dj

tubeme, 2016-10-11:
We don't want to lower the P gain as the plane starts to be slow responding to strong wind.

sanderux, 2016-10-11:
Your FW_RR_I is much to heigh, try 0.01 first and move up from there

sanderux, 2016-10-11:
Here's a comparison between our parameters:
http://params.droneslab.com/?name%5B%5D=vasil&log%5B%5D=http%3A%2F%2Flogs.uaventure.com%2Fview%2FszHmDnbEvtBKnGB8bfx4Dj&name%5B%5D=deltaquad&log%5B%5D=http%3A%2F%2Flogs.uaventure.com%2Fview%2FQEvHjGSwGTVceBTzk4uD47&defaultBase=master&compare=compare

tubeme, 2016-10-11:
I will try this and see. But I'm pretty sure it will degrade the performance in strong wind.

kd0aij, 2016-10-11:
Seems like there should be parameters for max roll and pitch rates in FW mode

tubeme, 2016-10-11:
There are parameters for max roll and pitch like FW_R_RMAX.

There should be parameters for max roll, pitch and yaw rate for Auto FW mode this includes the FW VTOL mode.

AndreasAntener, 2016-10-11:
What is actually needed is input shaping in the FW position controller.

kd0aij, 2016-10-11:
@tubeme PR https://github.com/PX4/Firmware/pull/5642 is a very simple change which constrains FW RPY rates to about 90 deg/sec, if you'd like to see whether that solves your issues.

Off topic: Looking at that code, I see that the "trim" values for roll pitch and yaw appear to be in "actuator" space, since they are added to _actuators.control[i] as the last step. I guess this means that, for FW, when you use QGC's "copy trims", the trim values expected are from flying in "manual" mode. Makes sense, now that I think about it.

kd0aij, 2016-10-12:
@tubeme Forget that PR. Instead, look at parameter FW_R_RMAX, which is 70 deg/sec in your log. That's not very fast, in my opinion, so I don't see why it would be a problem, but you could try reducing it.

tubeme, 2016-10-13:
The problem reducing FW_R_RMAX is that we make the plane very sluggish. 

We are aiming at operational conditions 15+ m/s In this kind of winds every rate limiting will mean inoperable plane. 

To tell you the truth we love very much how the plane is flying... The only problem is it is banging on the sticks during mission turns. We'd love to have something like the FF gain but only for Autonomous modes. With which we will make the mission commands more softer if we need so.

tubeme, 2016-10-13:
Why there was the need for implementing MC_YAWRAUTO_MAX?? Probably the same reason as ours but for quads and back in time.

kd0aij, 2016-10-13:
@tubeme FW_R_RMAX **does** control maximum roll rate, but it does not limit the maximum roll angle. So I don't see why it would prevent you from flying in wind or turbulence. I've never seen turbulence flip a plane, not even light foam slope soarers, they just bounce around the commanded roll angle.

Your log shows the roll rate is overshooting the setpoint of 70 deg/sec, peaking at 90 deg/sec. That implies the roll P gain may be a bit too high, or that you need some damping. But the yaw rate is oscillating even more; have you tried reducing the yaw P gain? 

You mentioned experimenting with the turn coordination method; the "closed loop" method is something I was working on a year ago with a 3D aerobatic plane, and I don't think the code that wound up in master is the same as what I flight tested, so you should stick with the default method which uses well-tested equations to calculate the required amount of rudder. (but I don't know how to tune that)
This is assuming that ECL_YawController is still handling yaw; I see that under some circumstances, a new algorithm called ECL_WheelController will be used instead, but I have no idea what it does.

tubeme, 2016-10-14:
In strong winds, it is important that the reactions are very fast in order for the plane to follow missions etc. At 7 m/s we can kite surf. At 15 m/s only the advanced kite surfers could go in the wind. So 15 m/s wind is a pretty strong wind. If we have a sloppy plane in 15 m/s it will most likely bounce around as you put it and will not be able to follow route. Out max speed is at 20-22 m/s so in this winds we have very little throttle envelope left.

Any way we will try to reduce some gains and see what is going to happen.

I'm puzzled by another thing. You say we have a Roll rate limited to 70 deg/sec but the plane is peaking at 90 deg/sec. This implies that there is no real limiting to 70 deg/sec. Why is that??

Lastly for the VTOL we have no YAW control during FW because the plane is Delta Wing.

The YAW example I gave was with another plane in another time. Just wanted to give another example but for YAW handling.

sanderux, 2016-10-14:
@tubeme what is the weight of your x8?

kd0aij, 2016-10-14:
@tubeme FW_R_RMAX constrains the max rollrate setpoint, not the actual rate. The 20 deg/sec overshoot is a characteristic of the way the PID control loop is tuned. You can try to reduce the overshoot by reducing P gain or adding damping, as I said before. You can't have "fast reactions" without overshoot unless there is some damping in the system. 

Are you really observing that 90 deg/sec roll rates are stressing the plane? The log shows that g-forces on the plane drop to .5 g during the roll, and peak below 1.5 g during the turn.
[image]

tubeme, 2016-10-14:
@sanderux 2.3kg for now going down. And it is our own frame it is not x8.

@kd0aij It is stressing the frame, because there are cuts in the frame for the vtol motors and the frame becomes narrow close to the body. And this 1.5g are in just perfect sunny dead calm day. In wind condition things change drastically. 

I understood now the difference so we will try dampen the system and reduce the P gain. We will report back with results.

Thanks a lot again.

tubeme, 2016-10-14:
@kd0aij Sorry for the stupid question. You mean dampening in terms of FW_L1_DAMPING right? and the L1 period right?

kd0aij, 2016-10-14:
No, and I just noticed that there is no damping term in the FW PID control loop. I guess the only thing you can do is reduce FW_RR_P and FW_RR_FF. Maybe try reducing the FF (feedforward) gain first.

Isn't there a PID tuning guide for FW somewhere? I've been tuning multis for so long I've lost track of the FW tuning procedures.

tubeme, 2016-10-14:
The FF gain is OK as we like it.  The only thing we can do is reduce the P and I gain for the Roll. We will try this.

There is a Manual for FW PID tunning. http://px4.io/docs/fixedwing-pid-tuning-guide/

But I made a tuning to a lot of airplanes. And we are pretty good at tuning the planes. 

We tuned our VTOL during years of work, many times. During flight tests in different conditions we make some corrections here and there. Also a couple of times the PX4 evolved and had some algorithms changed and tuned, which led to another run of PID tuning. This how incrementally we reached our current PID values. And we like how it flyies with this values in very strong weather. It is just too tight and in plain conditions this creates another problems like our overshoot.

We will just loosen the PID's a bit. With the EKF2 things might get OK even in strong conditions. 

I suppose EKF2 is the default Pos Estimator for FW now? Before it was EKF and before it was INAV so we were more dependable on adequate Tight PID tuning for the FW flight.

kd0aij, 2016-10-14:
@tubeme Thanks for the pointer to the tuning guide. I see that there is no mention of the trim parameters (TRIM_ROLL, TRIM_PITCH, TRIM_YAW) or QGC's "copy trims" function. Do you ever do any trimming in manual mode, or do you always leave those params at zero? I assume you never fly the VTOL in manual mode, but what about your big high-wing plane?

tubeme, 2016-10-14:
The trims are for the RC and we seldom fly manual in FW. We use manual only to check if everything is ok with the frame and the CG if we have some doubts, something like a double check. And we don't trim it during this sessions. We gradually moved from flying Stabalized mostly to Position, Altitude and Mission Flying recently. 

The QGC's copy trims is recent development so no mention in the UM. I should correct this. Tahnks for the memo.

dagar, 2017-04-11:
I think we can close this? Can you open a fresh issue if you're still having problems?

closed 2017-04-11 by dagar