PX4/PX4-Autopilot #5638: uavcan build failed

haigithub, 2016-10-11:
I have cloned the newest Firmware repo, and have updated the submodules (env: win7_x64). When I build the project, it failed with the prompt message as follow:

make[1]: warning: -jN forced in submake: disabling jobserver mode.
[  0%] Built target git_gencpp
[  0%] Built target git_genmsg
[  0%] Built target git_uavcan
[  0%] Built target git_nuttx
[  0%] Built target git_mavlink
[  0%] Built target git_matrix
[  0%] Built target mixer_gen
[  0%] Running dsdl compiler
[  1%] Built target xml_gen
[  1%] Built target nuttx_export_bugoo
[  1%] Built target nuttx_export_px4io-v2
Compiler failure
Traceback (most recent call last):
  File "g:/enouav/aplan/Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_com
piler/libuavcan_dsdlc", line 61, in <module>
    dsdlc_run(args.source_dir, args.incdir, args.outdir)
  File "g:\enouav\aplan\Firmware\src\modules\uavcan\libuavcan\libuavcan\dsdl_com
piler\libuavcan_dsdl_compiler\__init__.py", line 62, in run
    run_generator(types, output_dir)
  File "g:\enouav\aplan\Firmware\src\modules\uavcan\libuavcan\libuavcan\dsdl_com
piler\libuavcan_dsdl_compiler\__init__.py", line 111, in run_generator
    die(ex)
  File "g:\enouav\aplan\Firmware\src\modules\uavcan\libuavcan\libuavcan\dsdl_com
piler\libuavcan_dsdl_compiler\__init__.py", line 89, in die
    raise DsdlCompilerException(str(text))
DsdlCompilerException: [Error 5] : u'g:\\enouav\\aplan\\Firmware\\src\\modules\\
uavcan\\libuavcan\\libuavcan\\include\\dsdlc_generated\\uavcan\\CoarseOrientatio
n.hpp'
[Error 5] : u'g:\\enouav\\aplan\\Firmware\\src\\modules\\uavcan\\libuavcan\\libu
avcan\\include\\dsdlc_generated\\uavcan\\CoarseOrientation.hpp'
make[3]: *** [libuavcan_dsdlc_run.stamp] Error 1
make[2]: *** [src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc.
dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  1%] Built target prebuild_targets
[  1%] Built target io_prebuild_targets
make[1]: *** [all] Error 2

julianoes, 2016-10-11:
Hi @haigithub, are you using the Bash on Windows or the old setup?

Generally, we don't support and recommend building on Windows, sorry about that.
http://dev.px4.io/starting-installing-windows.html

haigithub, 2016-10-11:
@julianoes  I am using "PX4 Toolchain Installer v14 for Windows Download" on Windows. I have an personal customed firmware version, and I developed that version for some time. Everything is ok. The build failing comes when I clone a new repo from https://github.com/PX4/Firmware.

julianoes, 2016-10-11:
Yes, as said, unfortunately you're on your own. We're not supporting Windows. If you can resolve it, we appreciate a PR but we don't have bandwidth to look into it.

haigithub, 2016-10-11:
@julianoes  Thanks! I think I have find the method to solve the problems. The method is as follows:
[image]
Then building succeeds.

closed 2016-10-11 by haigithub

julianoes, 2016-10-11:
@haigithub Oh great, that was easy then. Thanks for sharing.