PX4/PX4-Autopilot #5639: Level calibration right after mag calibration doesn't seem correct

julianoes, 2016-10-11:
I have observed twice that the first level calibration right after having done all sensor calibration doesn't correctly show the progress and also produces a suboptimal result, presumably it doesn't really run at all.
A subsequent level calibration will fix it. This happened on a Pixhawk and Pixracer.

```
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] front side done, rotate to a different side
INFO  [commander] [cal] front side done, rotate to a different side
WARN  [hmc5883] mag cal status changed 
INFO  [commander] [cal] mag #0 off: x:0.04 y:-0.05 z:-0.09 Ga
INFO  [commander] [cal] mag #0 scale: x:1.00 y:1.12 z:1.04
INFO  [commander] [cal] mag #1 off: x:0.02 y:-0.03 z:-0.27 Ga
INFO  [commander] [cal] mag #1 scale: x:1.00 y:1.00 z:1.00
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] calibration done: mag
INFO  [commander] [cal] calibration started: 2 level
WARN  [navigator] navigator: global position timeout
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] calibration done: level
INFO  [commander] [cal] calibration started: 2 level
INFO  [commander] [cal] progress <0>
WARN  [navigator] navigator: global position timeout
INFO  [commander] [cal] progress <9>
INFO  [commander] [cal] progress <18>
INFO  [commander] [cal] progress <27>
INFO  [commander] [cal] progress <36>
INFO  [commander] [cal] progress <45>
INFO  [commander] [cal] progress <54>
INFO  [commander] [cal] progress <63>
INFO  [commander] [cal] progress <72>
INFO  [commander] [cal] progress <81>
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] calibration done: level
```

dagar, 2017-04-12:
Is this still an issue?

julianoes, 2017-04-12:
Still the same, just tested with a pixracer:

First level cal straight after accel cal:
```
INFO  [commander] [cal] Hold still, measuring up side
INFO  [commander] [cal] up side result: [ -0.1978  -0.5740  10.1749]
INFO  [commander] [cal] progress <102>
INFO  [commander] [cal] up side done, rotate to a different side
INFO  [commander] [cal] up side done, rotate to a different side
INFO  [commander] found offset 0: x: -0.083642, y: -0.129229, z: 0.380216
INFO  [commander] found scale 0: x: 1.000255, y: 0.999604, z: 0.999480
INFO  [commander] found offset 1: x: -0.083158, y: 0.046675, z: 0.160954
INFO  [commander] found scale 1: x: 1.002068, y: 0.998735, z: 0.994991
INFO  [commander] [cal] calibration done: accel
INFO  [commander] [cal] calibration started: 2 level
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] calibration done: level
```

Then a second try of level cal: 
```
INFO  [commander] [cal] calibration started: 2 level
INFO  [commander] [cal] progress <0>
INFO  [commander] [cal] progress <9>
INFO  [commander] [cal] progress <18>
INFO  [commander] [cal] progress <27>
INFO  [commander] [cal] progress <36>
INFO  [commander] [cal] progress <45>
INFO  [commander] [cal] progress <54>
INFO  [commander] [cal] progress <63>
INFO  [commander] [cal] progress <73>
INFO  [commander] [cal] progress <82>
INFO  [commander] [cal] progress <100>
INFO  [commander] [cal] calibration done: level
```

julianoes, 2018-02-21:
I want to verify that one again.

julianoes, 2019-01-30:
Note to self: should try this again.

julianoes, 2019-02-13:
Bloody bots these days.

julianoes, 2019-07-01:
Guys, I have enough to do without bots talking to me.

julianoes, 2019-10-01:
Seriously...

julianoes, 2019-10-01:
I can't reproduce the problem anymore, closing.

closed 2019-10-01 by julianoes