PX4/PX4-Autopilot #5643: Stack overflow possible in mixer_multirotor.cpp

chungkim, 2016-10-12:
At line 128 in src/modules/systemlib/mixer/mixer_multirotor.cpp (master branch), sscanf can overflow the 8-byte local buffer geomname when the part of its first argument (buf) that matches "%s" is longer than 7 characters.  The if statement at line 133 checks the overflow, but it does not prevent some code from modifying the return address in the stack by overflowing buf. In such a case, function MultirotorMixer::from_text will return to somewhere else regardless of the check.

Maybe put the size in the %s conversion specifier?:

`if (sscanf(buf, "R: %7s %d %d %d %d%n", geomname, ...) != 5) {`

LorenzMeier, 2016-10-12:
@davids5 Can you have look?

davids5, 2016-10-12:
@chungkim - Thank you! Keep up the good work!

@LorenzMeier - The solution makes sense to avoid a overwrite if a mixer file is not to spec.

Should we go to a field width of 8 with 9 byte buffer?

closed 2016-10-12 by davids5