PX4/PX4-Autopilot #5645: NULL pointer dereference in sched_note_switch

chungkim, 2016-10-12:
At line 146 of src/modules/systemlib/cpuload.c (master branch), there are times when "system_load.tasks[i].tcb" is NULL but it is dereferenced by "system_load.tasks[i].tcb->pid".  This results in reading an arbitrary value from a low address (that is, the offset of pid).  This happens because sched_note_switch can be called by NuttX before cpuload_initialize_once is called to initialize system_load.tasks.  Tested with NuttX 7c5ef883fc3307b44bef9d5f7b51cb144cdf400a on Pixhawk PX4FMU-v2.

Adding the following check between lines 144 and 145 resolved the issue for me:

`if (system_load.tasks[i].tcb == NULL) continue;`

davids5, 2016-10-12:
@chungkim - Thank you for point this out! 

I do not see Nuttx @ 7c5ef883fc3307b44bef9d5f7b51cb144cdf400a  on PX4/Firmware @ master branch.

The best thing to do is a PR against master so it is all tied in.

FYI: I believe this is fixed in the nutt_v3 branch already.

chungkim, 2016-10-12:
Oops, it's my mistake.  Updated in the post.

chungkim, 2016-10-12:
Mode detailed information:

In my setup with PX4/NuttX @ 154289bbf80738c97e5c062729b6904e858c3cef, sched_note_switch is called by sched_addreadytorun (PX4NuttX/nuttx/sched/sched_addreadytorun.c:128) in the context of "hpwork" kernel thread (thread start function: work_hpthread) **before** cpuload_initialize_once is called by the "init" task (call path: nsh_main -> nsh_initialize -> nsh_archinitialize -> cpuload_initialize_once).

This happens because a number of kernel threads, including "hpwork", are created (from PX4NuttX/nuttx/sched/os_bringup.c:172) before the "init" task is created (at PX4NuttX/nuttx/sched/os_bringup.c:230).  And when those kernel threads are created, sched_addreadytorun (and cpuload_initialize_once) is called before cpuload_initialize_once is called by "init" to initialize system_load.tasks.

It looks like "init" is still the last task to be created in nuttx_v3:

https://github.com/PX4-NuttX/nuttx/blob/0bc19a63bbae434daee5aac3c45a6aa26857ebeb/sched/init/os_bringup.c#L292

But I am not sure if this still exists in nuttx_v3.  Maybe somebody can simply check if sched_note_switch is called before cpuload_initialize_once in nuttx_v3?

Or should I report it to NuttX people?  I reported it here first because I thought checking the validity of system_load.tasks before use is the safest way to avoid a problem regardless of the underlying NuttX design. :-)

closed 2016-10-12 by LorenzMeier

davids5, 2016-10-12:
Just an fyi 
Upsteam NuttX change the interface to sched_note. While the operation is differnent the same start up issue existed and was ficed here: 
https://github.com/PX4/Firmware/commit/dae49ec69abc33268dc618927cbf995f08503594#diff-42ffbc0836b61b18a8d60a17849db9ae

commit 569cceb0 (2016-10-12): Fix task load measurement in NuttX. Reported by @chungkim. Fixes #5645.