PX4/PX4-Autopilot #5649: Problem Using D-GPS

meteor0530, 2016-10-12:
Hi

I updated the PX4 firmware through QGC 3.0.1. I used INAV and D-GPS for position estimate, and used a external compass. when I switch to POSCTRL using RC, the multirotor started to rotate around a point and the radius increased slowly. However, when I used the same compass with a GPS, everything went well. Does it mean D-GPS cant be used in px4?

Any help is appreciated.

julianoes, 2016-10-12:
Hi, can you please elaborate what kind of D-GPS you're using?
@bkueng can probably help better.

meteor0530, 2016-10-12:
@julianoes 
OEM615 using GPS-RTK

bkueng, 2016-10-12:
@meteor0530 does it provide the same output format in RTK mode? If not, it is currently not compatible.
Can you share a log?

meteor0530, 2016-10-13:
Both GPS and D-GPS provide GPGGA format
when I used D-GPS, I monitored the position in mavlink inspector and it was nice

meteor0530, 2016-10-13:
@julianoes @bkueng 
Can PX4 use double GPS for localization?

julianoes, 2016-10-13:
What do you mean with "double GPS". PX4 supports interfacing to two GPS modules. However, I don't think that has much to do with D-GPS or RTK.

bkueng, 2016-10-13:
So the GPS driver and communication seems to work then. It seems your problem lies elsewhere and you need to share a flight log for us be able to help you.

meteor0530, 2016-10-14:
@bkueng 
how to upload a .px4log file?

julianoes, 2016-10-14:
http://logs.uaventure.com/

meteor0530, 2016-10-14:
already uploaded, waiting for reply. thanks!

LorenzMeier, 2016-10-23:
Closing due to no further feedback.

closed 2016-10-23 by LorenzMeier