PX4/PX4-Autopilot #5650: Move CI to arm-none-eabi-gcc to 5.4.1

julianoes, 2016-10-12:
I think we should move the arm-none-eabi toolchain to the same version as used with the Ubuntu 16.04 installation which is 5.4.1. Currently, it's still on 4.8.4, see https://travis-ci.org/PX4/Firmware/jobs/166972099#L449, and https://semaphoreci.com/px4/firmware/branches/master/builds/512.

The installation [recommendation in the Devguide](https://github.com/PX4/Devguide/blob/master/starting-installing-linux.md#nuttx-based-hardware) is 5.4.1 from the [team-gcc-arm-embedded ppa](
https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa?field.series_filter=xenial).

dagar, 2016-10-13:
That seems like quite a jump, but we might as well go for it. @LorenzMeier talked about 1.5 being a stable release before updating nuttx, etc. Should we play it safe for 1.5 and keep 4.8.4 for one last release?

davids5, 2016-10-13:
How does this sound.

Note - I have not vetted gcc 5.4.1 on all SoC (M7 etal)

For the update of nuttx we will have to move to gcc 4.9.3. 

So if the code is stable at 1.5 an and then the only change to gcc 4.9.3 and do a 1.5.1 release - that will reaffirm @dagar usage of 4.9.3 in flight. 

Then 1.5.1 -> nuttx_v3 is just a OS and code change not a tool change.

dagar, 2016-10-13:
That sounds good to me, I just think we need a working gdb.

davids5, 2016-10-13:
Yep -I Use 4.8.4 gdb with 4.9.3. But it would be sweet to get to the bottom of the seg fault.

dagar, 2016-10-13:
Is there another incremental 4.9 release we can try (with good packages)? It would be ideal to have gcc and gdb together given the manual install instructions.

davids5, 2016-10-13:
@dagar if we have to vet something new with all the SoCs we could try the version that @julianoes is suggesting. But we would have to build all the targets in nuttx_v3 and test it on all the SoCs. That is a lot of regression testing and at the moment I am not able to do that.

julianoes, 2016-10-13:
So @bkueng and I both have used versions > 5.x.x for a while now (several months if I'm not mistaken) without issues. We have not flown it extensively but done all the bench testing with it.

davids5, 2016-10-13:
@julianoes - that is good to know. Have either you or @bkueng built and tested all the target on nuttv_v3? Including the F7? 

Would you please do a build and run 'make sizes;

bkueng, 2016-10-13:
I use gcc 5.2.0 for ARM and px4fmu-v4 and gcc 6.2.1 on the host, and havn't run into any problems so far. But I agree with @davids5 that we should be careful and switch only one at a time, not both compiler and OS version.

nuttx_v3 currently fails for me with:
```
px4io-v2/NuttX/nuttx-export/include/nuttx/fs/ioctl.h:374:27: error: "/*" within comment [-Werror=comment]
 /* see nuttx/include/modem/*.h */
```

julianoes, 2016-10-13:
For me that point is that we should match the Devguide to what we build and ship with CI. If the consensus is to use 4.8, 4.9, or any particular version, we should also reflect that in the Devguide.

I don't have a strong opinion or any knowledge of the difference, I just used it because it was in that repo and convenient.

davids5, 2016-10-13:
@julianoes - got it!
@bkueng - hmm - is that a fresh pull and current Nuttx? There is a patch should be applied on older version that fixed that and on the latest it was fixed in upstream.

My "goto" script when switching from master to nuttx_v3 is 

```git submodule sync --recursive && git submodule update --init --recursive```

from nuttx_v3 to master

```rm -fr NuttX && git submodule sync --recursive && git submodule update --init --recursive```

The ```rm NuttX``` is needed because the repo layout is different from old NuttX and the latest upstream that uses submodules.

bkueng, 2016-10-13:
@davids5 Thanks, that helped. Compiles fine now. `make sizes`:
```
   text    data     bss     dec     hex filename
 928072    4548    9828  942448   e6170 build_px4fmu-v1_default/src/firmware/nuttx/firmware_nuttx
1012068    4828   19644 1036540   fd0fc build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx
1241732    5476   23588 1270796  13640c build_px4fmu-v4_default/src/firmware/nuttx/firmware_nuttx
```

` make px4nucleoF767ZI-v1_default` fails however:
```
  File "/home/beat/px4/src/Firmware_nuttx_next/cmake/nuttx/bin_to_obj.py", line 52, in <module>
    locals())
  File "/home/beat/px4/src/Firmware_nuttx_next/cmake/nuttx/bin_to_obj.py", line 47, in run_cmd
    raise RuntimeError(stderr)
RuntimeError: Assembler messages:
Error: unknown cpu `cortex-m7'
Error: unrecognized option -mcpu=cortex-m7
```

bkueng, 2016-10-13:
I did some googling and the m7 error is a bug in gcc 5.2.0: https://bugs.archlinux.org/task/46951

davids5, 2016-10-13:
@bkueng 

Thank you!

LOL so much for for using the built-in package....

What do you feel our best option is?

bkueng, 2016-10-13:
Haha, true! Actually I'm using the built-in fedora package which also happens to be 5.2.0.

I think you can make the better call on this since you have far more experience than I have. In my opinion the risk of using a newer compiler is in that case not so high, as Julian and I have been using it (and some others as well), but we should at least do some testing for all supported platforms, so we could switch soon, and then focus on the nuttx upgrade which requires more in-depth testing.

davids5, 2016-10-14:
@julianoes , @bkueng 

It seems we are between as rock and a hardplace. I think we should try to determine the root cause of the fault in 4.9.3 gdb with our build. If it can be eliminated easily, then  we have a vetted solution in 4.9.3.

@LorenzMeier - What would like to see done here?

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
```

davids5, 2016-10-21:
@dagar @julianoes, @bkueng 

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

It builds F7 and Debugger works on F4 ! :)

closed 2016-10-21 by julianoes

bkueng, 2016-10-21:
Just to add as a reference here: I tested arm GCC 6.1.0 (which is in my testing repo) and after fixing some compiler Errors, I got this:
- Pixracer aka fmu-v4 builds and seems to boot & run properly
- Pixhawk aka fmu-v2 builds but gets stuck on boot with this output:
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
```

davids5, 2016-10-21:
@bkueng 

Good data point. Will you be debugging it?

We should also add the ```make sizes``` from nuttx_v3 Makefile to master and the check targets to start getting bin file size data.

bkueng, 2016-10-22:
Would be an interesting task, but as it could end up being a much bigger task than expected, I don't plan to do it at the moment.

davids5, 2016-10-22:
Understood. Please give me a link to the version of the compiler you are using or the --version output and I will take a look when time permits.

bkueng, 2016-10-24:
Here you go: https://bodhi.fedoraproject.org/updates/FEDORA-2016-ff04fe5b97
And the binary downloads:
http://koji.fedoraproject.org/koji/buildinfo?buildID=778715
http://koji.fedoraproject.org/koji/buildinfo?buildID=776054
http://koji.fedoraproject.org/koji/buildinfo?buildID=777227

yangang123, 2017-03-14:
yes