PX4/PX4-Autopilot #5651: X Plane HIL unable to work

ruyi7952, 2016-10-12:
HW: PixHawk 2.4.7  CUAV Clone
FW: Standard
X-Plane 9.70 / 10.51r1

I have read http://www.pixhawk.com/users/hil to do this one.

BTW,No Throttle No Ailerion No Elevator No Rudder in X-Plane， I can ben sure my RC  on air ,The ground station is correct 。

why?

dagar, 2016-10-13:
Has this ever worked for you? Did you select the HIL plane airframe? When you press connect in QGC does the map location match xplane? Are you able to arm from RC?

ruyi7952, 2016-10-13:
Airframe: SIM-XPlane ,
QGC location ok, with xplane 10.51r1 , XPlane 10.45/10.50 map error,
Yes arm 。

dagar, 2017-02-16:
@ruyi7952 can you try this again with latest PX4 and QGC?

zthorson, 2017-02-17:
I am seeing the same issue attempting an HIL simulation using a pixhawk board with no sensors, actuators, or RC connect.  I have disabled all circuit breaker checks, which allows me to arm.  This is for a fixed wing simulation.

**Symptoms**

* Similar to ruyi7952. 
* XPlane 10 packets at ~30 Hz
* Location in XPlane matches QGroundcontrol.  Simulated attitude and gps location match as well
* Enable sensor level HIL and USe newer actuator format is checked
* Arm/Disarm commands are accepted
* Setting up a mission and arming results in messages saying 'starting missions' and 'taking off' but no motion is shown in QGroundControl or XPlane
* If I manually fly the plane in XPlane, the location tracks correctly in QGround control.  
    * The link just doesn't appear to work the other way (QGroundControl managing flight)
    * 'Use newer actuator format' must be unchecked for me to do this, otherwise something is fighting and pushing my throttle back to 0.  This makes me think QGroundcontrol has throttle control, but for some reason isn't using it at the start of a mission.
* Using virtual joystick and *Manual* mode would not change the throttle setpoint in XPlane after arming the plane.  Though the joysticks could arm/disarm the motor.
* Attempting a manual 'Takeoff' mode command results in a few messages:
  * Info: Taking off
  * Critical: Using minimum takeoff altitude: 25.00 m
  * Critical: Takeoff denied, disarm and retry

**Hardware:**

* Pixhawk Based Board (Flight Tested and working)
* USB Data Connection
* No other connections to board

**Software/Firmware:**

* Windows 10 Platform
* XPlane 10.51r2 (build 105101 64-bit)
* XPlane Model: [HiLStar17](http://diydrones.com/profiles/blogs/hilstar17-public-release)
* QGroundControl: Tried 3.1.0, 3.1.3
* PX4 Firmware: Tried 1.5.5, 1.6.0rc1 and 1.6.0dev (master)

zthorson, 2017-02-23:
I found the issue.  It is very specific to PX4FMU_V4 hardware builds.  In this build, a workaround to flaky boards was put in place in https://github.com/PX4/Firmware/commit/021f0840ae7f724bfdfd317dda3d57362017b835

Basically, if you are using Pixracer (or variants thereof), when you attempt to enter HIL mode, the code meant to reset the sensors leaves the fmu driver running.  This uses up the /dev/pwm_output0 file structure, preventing pwm_out_sim from registering to it, causing the simulated pwm not to run.  This resulted in the simulator not being able to control throttle or other control surfaces.

I am going to look into having the sensor_reset call to the px4fmu driver shut down and clean up after the lines have been reset.  I can't think of a case where you would want the sensor_reset command to leave the fmu driver running, since it gets fully configured in a later step of the rcS script anyway.

@ruyi7952 I will try to get a pull request in next week, but if you would like to try it on your hardware sooner, I can send you a firmware file with the changes.

ruyi7952, 2017-03-11:
Thanks for your reply. 
I am using 

Futaba T18SZ with firmware version 2.0,
X Plane 10.51r2,
QGroundControl Tried 3.1.3, 
PX4 Firmware: 1.6.0dev (Master), 
PX4FMU_V2 (not PX4FMU_v4) (MZ). 

my computer runing windows 7 pro sp1 and macos 10.12.
The same problem, no progress.

fernandourrets, 2017-04-23:
Hello!!!! I've the same problem's Ruyi7952!!!! Please help me out!!!!! I'm working three days ago!!!! I'm so tired, please!!!!!!!!!!!!!!!! I don't want to give up!!!!! I just want to use the simulator!!!! Thanks a lot for your answer. Fernando

dagar, 2017-04-23:
@fernandourrets have you tried PX4 master?

fernandourrets, 2017-04-24:
Hello Dagar!!! Nice to meet you, not really, where can i download that Firmware???

fernandourrets, 2017-04-24:
I can Arm from my RC in the HIL but the throtlle, yaw, roll and pitch doesn't respond and if i make a Mavlink inspector the RC channels responding very well!!!! thks again

dagar, 2017-04-24:
First make sure you have the latest QGroundControl (http://qgroundcontrol.com/downloads/). Then go to the firmware tab under settings to flash the latest PX4, except select Advanced and look for master.

fernandourrets, 2017-04-24:
The Qgroundcontrol version is v3.1.3 and the firmware that i have available is PX4 standard version (stable), beta testing (beta) or Developer Build (master)

fernandourrets, 2017-04-24:
Done!!!! The master firmware is loaded....1) i must to calibrate all the sensors with Plane airframe and then switch to Simulation Hilstar (xplane)??????

fernandourrets, 2017-04-24:
I made everything with Master firmware but every time the accel, gyro, mag, etc became descalibrated, i can't do nothing!!!! Hellllllllp!!! Thanks

fernandourrets, 2017-04-24:
I must to go back and reload the Beta Firmware!!!! I can Arm by my RC Spectrum DX7 but doesn't respond the throttle, yaw, roll and pitch, if i select the Analyze function in the Widgets, select the rc channels and responds ok!!!!! I don't know what to do!!!!! Thanks again

TheRibbonRed, 2017-08-23:
@fernandourrets if you're still after HIL, you might wanna check out PX4 Flight Stack version 1.5.5. It works completely with HIL. It seems the ver 1.6.5 has another HIL bug in it that's yet to be resolved.

You can find the previous versions of PX4 Flight Stack on its GitHub:
https://github.com/PX4/Firmware/releases

fernandourrets, 2017-08-23:
Thank you so much!!!! I'll check out this version and then will tell you if
works!! Fernando

1MSG1, 2017-08-30:
@TheRibbonRed confirming that it's still inoperable. I'm able to arm (after lots and lots and lots of finessing it) but initiating a sequence in QGCS is impossible. Throttle locks to 50% and control surfaces don't respond to inputs.

TheRibbonRed, 2017-09-01:
@1MSG1 did you try loading the PX4 Flight Stack version 1.5.5 manually? Via "custom firmware" option? The PX4 team already made the .px4 firmware and zipped it, all you have to do is download from the GitHub and choose the firmware that suits your board.

As far as I know, the 1.6.5 version still has HIL bugs.

1MSG1, 2017-09-02:
Yes, tried that and it worked. For context I spent three days going mad
trying to figure out what I was doing wrong - it's 100% worth a note
somewhere (without having to dig through pull requests) that HIL on the
latest version is inoperable - really frustrating that it wasn't documented
in a way that's standard.

macfly1202, 2017-09-13:
Hi, I'm still looking for HIL with last px4 firmware. Same issue as describes above.
I successfully run HIL mode with PX4 1.5.5 but mission loading doesn't work aka activating "mission" after waypoint sync put the HIL UAVin "land" mode. 
Does anybody know the last "more" complete/running firmware version of Hil mode ?

thanks

1MSG1, 2017-09-13:
Hey Jean -

I gave up on and moved over to Q Ground Control which works ok in HIL.

1MSG1, 2017-09-13:
> @1MSG1 <https://github.com/1msg1> : I'm speaking about qgroundcontrol :
> it's what I use ! Have you sucessfully load waypoint/run mission from
> qgroundcontrol ?

If you revert back to the older firmware does that solve the issue? Put
differently, have you tried to flash a new SD card with the older firmware?

macfly1202, 2017-09-15:
Sorry for late answer - miss message. 
So I successfully run HIL mode with PX4 1.5.5 but mission loading doesn't
 work aka activating "mission" after waypoint sync put the HIL UAVin "land"
 mode.
What do you mean with "tried to flash a new SD card with the older firmware?"
I flashed my pixfalcon with px4 1.5.5 and Hil basic command works with X plane 10 but impossible to launch a mission after sync : go directly to landing.

fernandourrets, 2017-10-13:
Hello!!!! I'm sorry for my delay, i can't find the "PX4 Flight Stack
version 1.5.5 firmware" , i thank you so much if you could put the firmware
link on Github!!!!! Thank you so much. Fernando

fernandourrets, 2017-10-17:
Hello TheRibbonRed!!!! Firstly thank you so much for your answer, could you
send me by email the link to download the  "PX4 Flight Stack version 1.5.5
firmware"???? I can't find it, thank you again. Fernando

macfly1202, 2017-10-17:
1.5.5 is directly access with QGroundControl . In load firmware section choose ´advanced mode’ and load 1.5.5?instead of latest. Really simple. I don’t remenber exact term but I think it’s advanced mode... on the right below the standard px4 firmware choose menu . Hope it help!

fernandourrets, 2017-10-18:
thanks for answer this email. I send attached by email the pic that i touch
when i want to upgrade the firmware. Doesn't appear 1.5.5 option. Thanks
again, Fernando from Córdoba Argentina

macfly1202, 2017-10-24:
Hi, I check this morning And seem that I mistake. You true : you could download px4 firmware 1.5.5 here: https://github.com/PX4/Firmware/releases. Choose the *.zip, download it, unzip and load as  Custom firmware.

fernandourrets, 2017-10-25:
Thanks a lot macfly1202!!!!!! It works!!!! i'm fighting with other things in QGC but the firmware issue is done!!!! Thanks again. Fernando

macfly1202, 2017-10-25:
@Fernando, I’m happy for you!
If px4 dev team could solve the HIL issue in latest release, it will be nice.

fernandourrets, 2017-10-28:
Hi everyone once again!!!!!! I could use Qground Control on Hil simulator
mode with Xplane 10, i have setup on my RC three flight modes...1-manual
mode 2- mission mode and 3- stabilize mode, when i take off in manual mode
the plane fly almost good...but finally fly (the engine or propeller
sometimes turn off) then when i switch to mission mode by my rc the
qgroungcontrol display and the voice say...mission flight mode and the
plane goes to the first waypoint until the plane in the xplane10 doesn't
respond and crash!!!!!! Have you got any idea???? Summary...the
qgroundcontrol responds good but not the xplane 10 in mission mode!!!! Thks
again. Fernando!!!!

macfly1202, 2018-03-13:
Hi, I make a short trial again with qgroundcontrol 3.3.0 and PX4 latest stable. HIL still broken, no RC command. I try the old firmware 1.5.5 and wasn't working anymore. 
Everything seems to be ok - arming ok - gps ok but I ve got no RC command it seems...Thottle, gyro do nothing. My hardware is CUAV Pixhack Nano with px4v4 fmu.  I check if joystick was activate instead RC command but this wasn't the case. Qgroundcontrol display some error incompatibility between Qground and firmware when dsiplay safety menu... 1.5.5 is really really old.

Why using HIL and Xplane ?
I want to remind me the fly mode and make a remind in memory of mission in HIL before go flying - it's why I want to use X plane and HIL... It's sad that HIL still not working with Xplane because it's simple for windows plateform. For sure I could go to linux and run Gazebo but it's too complicated for simply play with flight mode and mission. 
I'm really use to do this with Arduplane and SITL and it's work well. 

Will dig further soon with other hardware....

LorenzMeier, 2018-03-13:
@dagar Are you using this setup?

macfly1202, 2018-03-13:
Thanks LorenzMeier. I test with Pixfalcon, PX4 FMU V2 1.5.5 stable, perfectly working with hilstar. Will test with latest build stable on the hardware now.

macfly1202, 2018-03-13:
I confirm even on PX4 FMU V2, latest stable build have no RC command in HIL mode.
I haven't V1(is it still exist?) and V3 to test...  

So only PX4 V2 FMU on 1.5.5 is working as now. 

I love to test, seriously, don't hesitate to ask me ! 
Thanks !

macfly1202, 2018-03-14:
fyi PX4 V2 FMU is sucessfully tested with xplane 11, like 10, same setup. Hil config Receive rate reach 37Hz (max), avg 31Hz on my config (versus 19 Hz max on xplane 10).
[image]

acfloria, 2018-03-15:
The problem is that the mixer is missing. The mixer was accidentally deleted just before the v1.6.4 release. I added it again but that was after the last stable release. So you either should use the master branch or add that [mixer](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/mixers/AERT.main.mix) to the code, build and flash then that version. In our group we are using X-Plane for HITL simulation. On a good enough computer it will run but you will need data at about 100 Hz otherwise the estimator will complain. To increase the data rate you should decrease the graphics options (such as visibility, cloud details,....) in X-Plane and deactivate vsynch in your graphics driver.

macfly1202, 2018-03-15:
Congrats for reactivity @acfloria  
Ok with travis master build at this day, it **RC command are working on px4 v4** now. 
About Xplane, I'm using all max details and option (:-)) in general ... 
I will try to desactivate option like you suggest but normally I have a quick powerfull graphic card I use for Deep Learning : nvidia 1080ti with suffisant ram (64gb) and 6 cores dual @ 3.2Ghz ... It must be sufficiant. Graphic renderer send back info to qgroundcontrol as you seem to suggest ?I have a look here https://dev.px4.io/en/simulation/hitl.html trying to understand what is slowing the system : UDP  ?  Will test you suggestions.
Thanks a lot !

macfly1202, 2018-03-15:
Another feed back : RC command are working on px4 v4 under HIL (master dev build) ...but there is no stabilization mode. All stabilized and auto mode crash plane (stabilzed, position, altitude, ). Like if there is no gyro. Checkbox 'Enable sensor level HIL' and 'Use newer actuator format' is checked. Same setup Rc link, xplane10 on px4 v2 stable haven't got this issue and perfectly working. So there is still an issue. sorry about that.

acfloria, 2018-03-15:
Can you check if acc/gyro data arrives on PX4 and the estimator converged? In addition what is the data rate you get currently?

macfly1202, 2018-03-15:
There is some data. cf screenshot. I redo a RC calibration (after tested beta build that wasn't ok because asking sensor calibration) and that seem to have fight the major issue. But there is still a big lag. My major concern is more than it's perfectly working with px4v2 and behaviour really different on px4v4. 

I'm not PX4 devops so do my best to answer your question. 

accel
[image]

gyro
[image]

Rate seem good > 100 
[image]

Qgroundlogfile *.log of this sensors during short xplane session with manual to stabilized mode
[20180315_13h35_px4fmuxplane_issue.log](https://github.com/PX4/Firmware/files/1815373/20180315_13h35_px4fmuxplane_issue.log)

acfloria, 2018-03-15:
Actually the log file from the pixhawk would help more than the QGroundlogfile. One thing I can spot is the `Mag0 fail Timeout` warning which most probably is pointing out an issue of the estimator.

I don't have a v4 on hand so I do not know why it might work worse than v2.

macfly1202, 2018-03-15:
Could it be hardware related (my CUAV pixhack nano) ? 

Here the log extract from SD for this ession (normally, it's not my local time log so not sure but seem 
 to time of Xplane offset, I didn't notice before they are linked) 

[log_0_2018-2-15-06-02-22.zip](https://github.com/PX4/Firmware/files/1815423/log_0_2018-2-15-06-02-22.zip)

acfloria, 2018-03-15:
On the plane the data only arrives at around 25 Hz, this definitely will cause the estimator to diverge. I suspect that it is an issue of QGC not relaying the data fast enough and producing dropouts. At least that happens when in our case when HIL fails. So I think at least the issue is not caused by your hardware alone although it could contribute to it a little as you say for v2 it works better.

macfly1202, 2018-03-15:
Log file for this error: this time logs are ok

https://logs.px4.io/plot_app?log=bf2ce9c1-6ea5-4bcc-8155-a29de77da9dc

[log_0_UnknownDate.zip](https://github.com/PX4/Firmware/files/1815527/log_0_UnknownDate.zip)

macfly1202, 2018-03-15:
Log (sucessful simulation) file for Pixfalcon px4 fmu v2 with Xplane and Hilstar 

http://logs.uaventure.com/view/hhVUmEjhf27j6XfnNw6BMo

[log_0_UnknownDate.zip](https://github.com/PX4/Firmware/files/1815665/log_0_UnknownDate.zip)

fernandourrets, 2018-03-18:
Hi, firstly i'm sorry my delay!!!!! I tried and it works using Qground
Control software / Simulation Plane / HilStar (xlpane), Firmware 15.5

If you want i can send to you the features on xplane 10 simulation software

I hope has been helpful!!!! Fernando

macfly1202, 2018-03-18:
Hi @fernandourrets, no problem -:)
When you say « it’s works » you mean Position and Altitude mode are ok with px4v4 and latest build ? because it’s the problem. The simulation is ok in manual mode for latest build on px4 fmu v4 but error with gyro/imu. 
About 1.5.5 and px4v2, I agree it’s perfectly working. 
Which firmware variants have you test ? V2 or v4 ?

Dohyeung, 2018-05-11:
Hello, I am trying to use Pixhawk, QGroundcontrol, XPlane10, and Easystar to make HITL testbed. I can figure out that I cannot use the latest version firmware in order to establish the testbed. Therefore, I am trying to update the 1.5.5 version firmware. On the QGC program, what files should I choose? When  I selected "nuttx-tap-v1-default.px4" file, there is an error like following

Downloading firmware...
From: /Users/dairum/Desktop/2018 Summer/DP project/Programs/Firmwares/Firmware/nuttx-tap-v1-default.px4
Download complete
Downloaded firmware board id does not match hardware board id: 64 != 255
Image load failed
Upgrade cancelled

In advance, thank you.

Dohyeung, 2018-05-15:
Hello everyone,

I have a quick question. I am using EasyStar II, PX4 FMU V2, XPlane10, QGC and firmware 1.5.5 for HIL testbed. Then is there any way to generate RC command for control the EasyStar II in HIL mode. In advance, thank you.

1MSG1, 2018-05-15:
> @Dohyeung <https://github.com/Dohyeung> YOu can connect an RC controller
> or use a joysticks/gamepads (RC will convert positions from that into
> MAVLink commands: https://dev.px4.io/en/simulation/#joystickgamepad-
> integration )

Dohyeung- Regarding your question on updating from 1.5.5, I believe that
1.5.5 is the latest availalbe version in which HIL works. Does anyone have
any insight on when HIL will be restored in the latest versions?

Dohyeung, 2018-05-15:
@hamishwillee Thank you so much. And I have one more question. In the HIL testbed, I think the PX4 FMU does not generate PWM signal for actuators. Am I correct? If I am right, could you tell me how I can activate the PWM signal for actual motors?

Dohyeung, 2018-05-15:
@1MSG1 Thank you. 1.5.5 is the available version for HIL testbed.

1MSG1, 2018-05-15:
Yes, thanks. My question is whether there's a plan to include HIL
capability in the latest versions (beyond 1.5.5). Also, I have the same
question as you regarding PWM signal.

hamishwillee, 2018-05-15:
@Dohyeung @1MSG1  What makes you think that HIL is not supported? It is most definitely "supported" for a number of simulators in the current version - https://dev.px4.io/en/simulation/hitl.html#compatible_airframe

That is not to say that it works, or works particularly well. That could be a bug, in which case you would post an [issue in Firmware](https://dev.px4.io/en/contribute/#have-a-problem)

> Thank you so much. And I have one more question. In the HIL testbed, I think the PX4 FMU does not generate PWM signal for actuators of the EasyStar II when I am just using the waypoints to control the flight. Am I correct? If I am right, could you tell me how I can activate the PWM signal for actual motors?

I do not know - I suspect that most actuators are supposed to be disabled on actual hardware and their movement sent to the simulator. The best place to ask this is either the discussion boards or Slack (#hitl channel).

fernandourrets, 2018-05-16:
Hi everyone, sorry for my delay answer!!!! I'll describe my configuration
witch works for simulation mode!!!!

Software....Qgroundcontrol + Xplane 10

Configuration in QGC...

1- Hil configuration: host 127.0.0.1:49000,
2- vehicle setup/airframe: simulation plane,
3- vehicle: hilstar (xplane),
4- firmware: 1.5.5

Configuration in Xplane 10...

1- net connections/ data: 127.0.0.1:49005

2- (check) ip of data receiver (for data input...)

3- configuration / setup

(check)        (first from the left and last one in the end) you have 4
fields to check, check only de first one and the last one for the below
fields numbers...

3, 4, 6, 16, 17, 19 20 and 21

4- udp velocity 96.0 / sec

That's all !!!! i hope i' ve been helpful. Fernando

Dohyeung, 2018-07-16:
@fernandourrets Hello fernandourrets. I am using XPlane10, QGC and firmware 1.5.5 for HIL testbed. It works well but I am trying to generate PWM signal to control actuators of Easystar II RC plane. Have you ever generated PWM signal to control actual motors of the plane?

Best regards.

fernandourrets, 2018-07-16:
hello, i'll answer you later, i'm at work right now. thanks

macfly1202, 2018-09-14:
> > fyi PX4 V2 FMU使用xplane 11进行了成功测试，如同10个相同的设置。Hil config接收速率达到37Hz（最大值），在我的配置上平均为31Hz（相对于xplane 10上的最大值为19 Hz）。
> > [image]
> > I want to  know that what kind plane is used in x-plane11 in hitl

using Hilstar17F from https://diydrones.com/profiles/blogs/hilstar17-public-release
Amazingly, this old plane have a nice behaviour in Xplane.

[1 more reply]

commit 63857c6a (2017-03-05): px4fmu-v4: Fix for HIL Unable to Set Control Surfaces. Fixes #5651

The sensor_rest command added to the startup script for px4fmu_v4 hardware
was leaving the /dev/pwm_output0 driver open.  This would prevent the
pwm_out_sim module from registering as a simulated driver.  The result
would cause the system to properly boot in HIL mode, but you would not
be able to set any control surfaces.

Since sensor_rest and peripheral_reset are only used on initialization, the
drivers can be shut down after they have performed their reset functions.
closed 2019-02-13 by stale[bot]